1technophile/OpenMQTTGateway #1804: LilyGo Lora 32 V2.1 with rtl_433 at 915Mhz

allistermaguire, 2023-11-15:
Hi, I am trying to use a LilyGo v2.1 with the rtl_433 to decode signals from my Cavius Smoke Detectors.

I know it isn't a documented configuration, and I have cobbled my configuration together from people's posts in the repo and the rtl_433_ESP repo and think I have a working configuration.

According to the Cavius spec sheet my Smoke detectors are using 926.365 MHz, they are using FSK_PULSE_PCM modulation which is supported in the latest version of rtl_433_ESP.

When I trigger the events with the Smoke Detectors, I can see the signal being picked up, it just isn't decoded correctly for some reason. I am not sure where to go from here?

It took me a while to get the Cavius devices working with an RTL-SDR, I finally worked out that the signal strength was too strong.

Here is the output, I have a forked version of rtl_433_esp where I make use of `MY_DEVICES` to free up memory. I am using another LilyGo for my EcoWitt devices and it is great thank you.

```
Reconnecting to /dev/cu.usbserial-56750030541 ................................   Connected!
N: ZwebUI setup done
T: Port: 8883
T: Mqtt server: xxx.xxx.xxx.xxx
T: ZgatewayRTL_433 command topic: home/OMG_lilygo_lora32_rtl_433_ESP/commands/MQTTtoRTL_433
N: ZgatewayRTL_433 setup done 
T: enableActiveReceiver: 3
N: Switching to RTL_433 Receiver: 926.36Mhz
rtl_433_ESP(6): SX1276 gpio receive pin: 32
rtl_433_ESP(6): SX1276 receive frequency: 926.364990
rtl_433_ESP(6): # of device(s) configured 3
rtl_433_ESP(6): ssizeof(r_device): 112
rtl_433_ESP(6): cfg->devices size: 336
Registering protocol [0] "Ambient Weather WH31E Thermo-Hygrometer Sensor, EcoWitt WH40 rain gauge"
Registering protocol [1] "Cavius smoke, heat and water detector"
Registering protocol [2] "Fine Offset Electronics WS80 weather station"
----- SX127x Status -----
RegOpMode: 0x0c
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x07
RegBitrateLsb: 0x40
RegRxBw: 0x12
RegAfcBw: 0x02
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----------- FSK --------------
FDEV_MSB: 0x02
FDEV_LSB: 0x8f
----- SX127x Status -----
T: mqtt_max_packet_size: 2560
N: OpenMQTTGateway modules: ["LilyGo_SSD1306","WebUI","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
N: Connected to broker

MQTT messages deleted here.

rtl_433_ESP(6): Signal length: 13278706, Gap length: 27440237, Signal RSSI: -39, train: 0, messageCount: 0, pulses: 347
pulse_slicer_pcm: Ambient Weather WH31E Thermo-Hygrometer Sensor, EcoWitt WH40 rain gauge codes [{3098}fcfbf3f3e7cf9f3e781f8003fe7fc001f9ffe7fc07fcf01f3e7cf9f7f3e7cfdf9f3e7fcf81ff3e07fc0f9f9ff03f7e7fc0ff81f3f3fe07efcf9f9f1f3f3fe7e7cf9f9f3e07fcf81f3fe07fc0f9f9f3e3e7e7cfcf9f9f3fe7c0ffbf03fc0f9fbfe07cf9ff03fe07cfcf9f1f3f3e7e7cfcf9f9fbf3e7cf9f3e7cfcf9f7f7cfcf9f3e7dfdf9f9f9f3e7cfbf9f3e7efcfbf3e7cfdf3fbf03e000ff3ff0003c7ff9ff03fefc0f9f9f3f3e7cf9f9f3e7cf9ff7e0ffcf81fe07cf9ff81f3e7fc0ff81f3f7fc0f9f9f3f7e7cf9ff3f3e7cf9f3e0ffcf83e7fc07fc1f9f9f7e7e7cf9f3efef9ff3f07fefc07fc1f9f9ff01e7cff80ff83f3e7cf9f3e7e7dfcf9f3f7e7cf9f3e7ef9f9f7f7e7e7cf9f3e7cf9fbe3e7cf9f3e7f7e7efe7df9f3e7cf9f3e7cf81f0007fcff0001f3ffc7fc0ff9f01f3e3e7e7cf9f3e7e7e7efeff9f81ff9e03fe0fcfcff80fbf3fe07fe0fdf9ff01e3efcf9f3c7cff3e7e7cf9f3e07f9f01f3fc0ff03e7cfdf9f3e7cf9f3f7e7fcf83ff3c07fc0f3fbfe03efeff80ff80fbe7cf800f8]
pulse_slicer_pcm: Exact bit width (in us) is 207.25 vs 206.00, 12 bit preamble
pulse_slicer_pcm: Exact bit width (in us) is 208.75 vs 206.00, 12 bit preamble
pulse_slicer_pcm: Exact bit width (in us) is 208.29 vs 206.00, 14 bit preamble
pulse_slicer_pcm: Exact bit width (in us) is 208.71 vs 206.00, 14 bit preamble
pulse_slicer_pcm: Cavius smoke, heat and water detector codes [{881}bb55531b0dd9a55757aba74cb67b32d9eb56dab534b32d56bbb4f997cae65aad6bb7aaaabdaafaaaeaf757721b85dcf2d6ab557dd32b9599796bd5b555d59b5aabdb7cdacacdaaababd55dbeaaabaaaeeeaaaa4361766956aab7d9d36d9b3bd97556aaa696657aaaed74cbcf99a88]
pulse_slicer_pcm: Fine Offset Electronics WS80 weather station codes [{3022}f9f7cf9f3e7cf9f3c0f8003fe7fc001f3ffcff81fe780f9f3e7cfbf3e7cfdf9f3e7fcf81ff3e07fc0f9f9fe07efcff81ff03e7e7fc0fdf9f3f3e3e7e7fcfcf9f3f3e7c0ff9f03e7fc0ff81f3f3e7c7cfcf9f9f3f3e7fcf81ff7e07f81f3f7fc0f9f3fe07fc0f9f9f3e3e7e7cfcf9f9f3f3f7e7cf9f3e7cf9f3e7dfdf3e7cf9f3efdf3e7cf9f3e7df9f3e7efcfbf3e7cfdf3f7c0f8003fcff8003c7ff9ff03fefc0f9f9f3f3e7cf9f9f3e7cf9ff7e0ff9f03fc0f9f3fe07cf9ff03fe07cfdff03e7e7cfdf9f3e7fcfcf9f3e7cf83fe7c1f3fc0ff07cf9f7cf9f3e7cfbf7cff9f83ff7e07f83e7cff81e7cff03fc1f3e7cf9f3e7cfbf3e7cfdf9f3e7cf9fbe7cfbfbf3e7cf9f3e7cf9fbe3e7cf9f3e7ef9f7f3ef9f3e7cf9f3e7cf81f0007fcff0003e7ff1ff03fe7c07cf8f9f9f3e7cf9f3e7dfdff3e0ffcf03fc1f3e7fc0fbe7fc0ffc1f7cff81e3ef9f3e78f9fe7cf9f3e7cf81fe7c0f9fe07f81f3e7efcf9f3e7cf9fbe7f9f07fe781fe079fbfc0fbf7fc07f80fbe7cf800f8]
rtl_433_ESP(6): # of messages decoded 0
```
Here is the same event via rtl_433 and a RTL-SDR device
```
% rtl_433 -f 926.365M -X "n=cavius,m=FSK_PCM,s=206,y=2700,l=206,r=1000" -g 5 -s 250k
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
[... 47 lines cut ...]
  '-DDISPLAY_IDLE_LOGO=true'
  '-DRF_SX1276="SX1276"'
  '-DRF_MODULE_INIT_STATUS=true'        ; Display transceiver config during startup
; *** Custom Settings ***
  '-DESPWifiManualSetup=true'           ; WiFi
  '-Dwifi_ssid="*******"'
  '-Dwifi_password="*******"'
  '-DMQTT_SECURE_DEFAULT=true'          ; MQTT
  '-DMQTT_CERT_VALIDATE_DEFAULT=false'
  '-DMQTT_USER="*******"'
  '-DMQTT_PASS="*******"'
  '-DMQTT_SERVER="xxx.xxx.xxx.xxx"'
  '-DMQTT_PORT="8883"'
  '-Dota_password="*******"'           ; WebUI Password
  '-UZmqttDiscovery'                    ; Disable HA discovery
  '-DOOK_MODULATION=false'              ; FSK mode
  '-DMY_DEVICES'                        ; Limit to specified devices
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'         ; Logging
  '-DDEMOD_DEBUG=true'
  #'-DRTL_ANALYZER=true'
  '-DRTL_DEBUG=4'                       ; Enable RTL_433 device decoder verbose mode for all device decoders ( 0=normal, 1=verbose, 2=verbose decoders, 3=debug decoders, 4=trace decoding. )
  '-DRTL_VERBOSE=1'                     ; Enable RTL_433 device decoder verbose mode, ## is the decoder # from the appropriate memcpy line in signalDecoder.cpp
custom_description = For ESP32, Gateway using RTL_433_ESP and RadioLib
custom_hardware = ESP32 LILYGO LoRa32 V2.1
```

1technophile, 2023-11-15:
Hello,

For information I'm working on a branch to enable the support of FSK(beta level) for the Lilygo/ttgo:
https://github.com/1technophile/OpenMQTTGateway/tree/fsk-modulation

If you want to try it. You can change the frequency in the webUI and do different tests easily this way 
[image]

allistermaguire, 2023-11-15:
Thanks @1technophile, I had a look and the only thing I see different is that I have this flag`'-DRF_SX1276="SX1276"'` to Enable support for SX1276 Transceiver. I have tried it without this flag and it made no difference.

I haven't actually looked at offsetting the frequency yet to maybe compensate for the cheaper ESP hardware, but this would make it much faster, thank you. I am not sure how to narrow the path forward from here, if it is simply that the radio is slightly off, or something else.

From a calibration perspective I know my RTL-SDR is good, and that I have the correct frequency for the device.
[image]

NorthernMan54, 2023-11-16:
@allistermaguire FSK Signal reception is pretty hit and miss for some people.   Did you see this tweak https://github.com/NorthernMan54/rtl_433_ESP/issues/87 ?  Does it work for you ?

allistermaguire, 2023-11-16:
Thanks @NorthernMan54, yeah I needed that to get my EcoWitt's working, but they are 433MHz and different LilyGo. Most things in NZ seem to use the 433 band over 915. I had it at 50 based on that, and I have tried the tuning values that you committed to master recently. Neither work.

I might see if I can find a known OOK signal around with the RTL-SDR, and then see what the LilyGo/Antenna combo looks like from a calibration perspective. Then I have a starting point to work off to start playing with other settings.

allistermaguire, 2023-12-04:
Hi @NorthernMan54, I got Ambient Weather WH31 in the 915Mhz range, and it works perfectly. The signal length from the WH31 is very short, whereas the Cavius smoke detector is much longer, up to 20 seconds.

NorthernMan54, 2023-12-04:
That is good to hear

closed 2024-03-04 by allistermaguire