1technophile/OpenMQTTGateway #1815: [FR] SJWS01LM support?

ilgrank, 2023-11-24:
Hi
As far as I can see, SJWS01LM  is not currently supported (but it seem to be supported in HASSIO)
[https://community.home-assistant.io/t/xiaomi-ble-support-for-water-leak-detectors-sjws01lm/483889](https://community.home-assistant.io/t/xiaomi-ble-support-for-water-leak-detectors-sjws01lm/483889)

..is there any plan to add support? Thanks!

DigiH, 2023-11-26:
Hi @ilgrank 

The SJWS01LM seems to only be supported by decrypting the encrypted BLE advertising data, along with the individual key.

https://custom-components.github.io/ble_monitor/by_brand#xiaomi

Decryption is currently not yet supported by OpenMQTTGateway, so it won't be possible to support the SJWS01LM until this has been added to OpenMQTTGateway.

If you find that there is also decodable advertising data, could you set [Advertisement and Advanced Data](https://docs.openmqttgateway.com/use/ble.html#advanced-advertisement-and-advanced-data-default-false) to true and post some samples from MQTT Explorer here with dry and wet states?

ilgrank, 2023-11-30:
> If you find that there is also decodable advertising data, could you set [Advertisement and Advanced Data](https://docs.openmqttgateway.com/use/ble.html#advanced-advertisement-and-advanced-data-default-false) to true and post some samples from MQTT Explorer here with dry and wet states?

While I can't provide a payload at the moment, Ircama, one of the contributors of pvvx/ATC_MiThermometerpvvx/Mithermometer, provided a PoC code for decrypting the payload [here](https://github.com/pvvx/ATC_MiThermometer/issues/410#issuecomment-1831595168)

DigiH, 2023-12-01:
The example given does require the bindkey to then be able to decrypt. Not possible with OpenMQTTGateway currently, but something we are looking into for the future.

Also, are you sure you have set [Advertisement and Advanced Data](https://docs.openmqttgateway.com/use/ble.html#advanced-advertisement-and-advanced-data-default-false) to true in your above example message? As there is no raw encrypted undecoded advertisement data showing in

`{"id":"18:C2:3C:23:xx:xx","UTCtime":"2023-11-30T23:29:44Z","unixtime":1701386984,"name":"Mi Flood Detector","rssi":-36} `

ilgrank, 2023-12-01:
Me bad, I had Advertisement and Advanced Data set to false. Turning it on, I get:
```
{
 "id":"18:C2:3C:23:xx:xx",
 "UTCtime":"2023-12-01T15:47:49Z",
 "unixtime":1701445669,
 "mac_type":0,
 "adv_type":0,
 "name":"Mi Flood Detector",
 "rssi":-56,
 "servicedata":"105963085ccb04233cc218",
 "servicedatauuid":"0xfe95"
}

```
pvvx has an online tool ([Telink Flasher 7.1](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)) to unbind these devices from Mi Home, but as far as I understand, that does not solve the need for a bindkey, right?

ilgrank, 2024-01-22:
..ehm, posting just to preven the issue going stale (sry) ':-)

DigiH, 2024-01-22:
I have to admit I haven't fully read the related issue through enough to know if or if not the Telink Flasher removes any need for a bindkey, but I somehow doubt it, if the actual device firmware is not being swapped out.

The best thing for you to do now is to post a few (5-10) messages as above with Advertisement and Advanced Data true, received at different times, when the sensor is dry, and then also 5-10 messages at different times when the sensor is wet.

This should enable us to see if there is a pattern for a decodable binary state for dry/wet.

closed 2024-05-07 by github-actions[bot]