1technophile/OpenMQTTGateway #1817: ESP32-C3 super mini not supported

michapr, 2023-11-27:
using the web upload I get the message:
"Your Unknown Chip board is not supported."

used airm2m_core_esp32c3 option and the suggested connection "USB JTAG/serial"

At first tried normal upload - failed, 
Then tried in bootloader mode and got the message that unknown chip (see above).

Board:
[image]

Any idea?

Thanks!

michapr, 2023-11-29:
> (BTW: the "lolin_c3_mini" is missing in options...)

was added in the meantime - thanks ;)
But using this web upload I get the same error.

BTW: The upload of the ready firmware is working (using the flash download tool) - so this is a way for all who want to use this chip.

BOlaerts, 2024-01-16:
In my experience flashing the new ESP32 S2/C3 boards works best with https://adafruit.github.io/Adafruit_WebSerial_ESPTool/

1technophile, 2024-01-16:
It is now supported in the development branch, accessible here https://docs.openmqttgateway.com/dev

https://community.openmqttgateway.com/t/install-on-esp32-c3-super-mini/2965/21

closed 2024-01-16 by 1technophile