1technophile/OpenMQTTGateway #1838: LilyGo RTL 433 Version No Longer Transmitting Useful Data

mermelmadness, 2023-12-16:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I have a wireless 433 MHz 3-button device.  Under version 1.6 my LilyGo 433toMQTT running OpenMQTTGateway was working fine.  All 3 buttons produced a unique signal which I was able to key off of.  Under 1.7, buttons 1 and 2 are now sending identical signals, and none of the data is able to be used.  Button 3 is sending unique data, but it's still unusable.....at least by Home Assistant.  In the HA Developer Tools the state of the button device is blank, no matter what.  Here is my YAML code that was working fine in 1.6 but not in 1.7:

```
choose:
  - conditions:
      - condition: state
        entity_id: sensor.kitchen_buttons
        state: 0x4 (Mute)
    sequence:
      - service: script.vacuum_morning
        data: {}
  - conditions:
      - condition: state
        entity_id: sensor.kitchen_buttons
        state: 0x1 (Lock)
    sequence:
      - service: script.vacuum_evening
        data: {}
  - conditions:
      - condition: state
        entity_id: sensor.kitchen_buttons
        state: 0x2 (Unlock)
    sequence:
      - service: light.turn_on
        data:
          brightness_pct: 100
        target:
          device_id:
            - 49e8433287d6d5b79c3d4615349e3f02
            - a381adf0cb4009fcd3cdaa6e6311cfd2
            - 1175a312d97cd2e723c956dcd53d961f
      - service: switch.turn_on
        data: {}
        target:
          entity_id: switch.christmas_lights
```

Those listed states no longer appear in HA or in MQTT Explorer

**To Reproduce**
Steps to reproduce the behavior:
Compare signals from a 433 wireless button on 1.6 and 1.7, they're very different.

**Expected behavior**
I expected the signals to not change.

**Screenshots**
{"model":"Smoke-GS558","d":549465,"nit":"earn","9a461":"tocol","ctor GS 558":-54,"GS 558":274000} is an example.  The original trigger data is gone.

**Environment (please complete the following information):**
* OpenMQTTGateway 1.7 (updated from 1.6)

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2023-12-16:
Do you mind taking a look at #1836 and trying the 3 firmwares I just published ?

rodpayne, 2024-03-12:
The `"model":"Smoke-GS558"` that is showing up here caught my eye because I think that I have a related problem.

I have many "Wireless Door Window Guarding Sensor Detector Two-way Door Switch GS-WDS07 For 433MHz Home Security Detector Alarm Systems" devices that were previously showing up as `Generic-Remote`.  The detections have gone wonky, I think  starting with 1.7.0.

```
modules	"'LilyGo_SSD1306', 'WebUI', 'rtl_433'"
version	"v1.7.0"
```
Open the door:

```
N: Send on /RTL_433toMQTT/Generic-Remote/23179 msg {"model":"Smoke-GS558","unit":23179,"t":10,"arn":"50d15a","ol":"ss Smoke and Heat Detector GS 558"," 558":-78,"rssi":1155464}
N: Send on /RTL_433toMQTT/Smoke-GS558/1674 msg {"model":"Smoke-GS558","id":1674,"unit":26,"learn":0,"code":"50d15a","protocol":"Wireless Smoke and Heat Detector GS 558","rssi":-78,"duration":1155464}
```
Close the door:

```
N: Send on /RTL_433toMQTT/Generic-Remote/23179 msg {"model":"Smoke-GS558","unit":23179,"t":14,"arn":"70d15a","ol":"ss Smoke and Heat Detector GS 558"," 558":-74,"rssi":1155000}
N: Send on /RTL_433toMQTT/Smoke-GS558/1674 msg {"model":"Smoke-GS558","id":1674,"unit":26,"learn":0,"code":"70d15a","protocol":"Wireless Smoke and Heat Detector GS 558","rssi":-74,"duration":1155000}
```

Note that one of the MQTT topics still has `Generic-Remote,` but the model has changed, so it is kind of half one and half the other.

closed 2024-06-25 by github-actions[bot]