1technophile/OpenMQTTGateway #1839: presenceUseBeaconUuid doesn't work anymore since V1.7.0

RoBra81, 2023-12-16:
**Describe the bug**
Since version 1.7.0 although presenceUseBeaconUuid is activated mac address will be send in MQTT presence message. Work find till version 1.6.0

**To Reproduce**
Steps to reproduce the behavior:
1. Enable HASS presence: mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"hasspresence":true}'
2. Enable presenceUseBeaconUuid: mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"presuseuuid":true}'
3. Check MQTT messages

**Expected behavior**
The uuid should be used as id within MQTT message

**Environment (please complete the following information):**
* OpenMQTTGateway version 1.7.0

absentwallaby, 2023-12-18:
I'm having the same issue.

absentwallaby, 2023-12-19:
ok - this is what I have set in 
".../MQTTtoBT/config"

{"presuseuuid":true,"pubuuid4topic":true,"presenceawaytimer":60000,"save":true,"minrssi":-10}

- Adding the minrssi doesn't seem to have helped - still no uuid / beacons showing up - only MAC addresses.

RoBra81, 2024-01-23:
Any news on this?

1technophile, 2024-01-26:
I will check it before releasing the new version

closed 2024-08-08 by github-actions[bot]