1technophile/OpenMQTTGateway #1840: Home Assistant autodiscovery creates devices with 'unknown' values

ZogKarndon, 2023-12-17:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Home Assistant autodiscovery creates devices with 'unknown' values.

I have several LaCrosse temperature/humidity sensors that get autodetected (yay) but the values in Home Assistant are always 'unknown'.   I suspect the autodiscovery mechanism for Home Assistant is using an incorrect template.

**To Reproduce**
Steps to reproduce the behavior:
1.  Configure OpenMQTTGateway and set up Home Assistant integration.
1. Turn on autodiscovery
2. Go to Home Assistant and look for new sensors.
3. Sensors appear with the expected entities, but all values are 'unknown'.

**Diagnostic info from MQTT Explorer**
MQTT topic; homeassistant/sensor/LaCrosse-TX141THBv2-0-102-temperature_C/config
value: 
```
{
	"stat_t": "+/+/RTL_433toMQTT/LaCrosse-TX141THBv2/0/102",
	"dev_cla": "temperature",
	"unit_of_meas": "°C",
	"name": "temperature",
	"uniq_id": "LaCrosse-TX141THBv2-0-102-temperature_C",
	"val_tpl": "{{ value_json.temperature_C | is_defined }}",
	"state_class": "measurement",
	"device": {
		"ids": [
			"LaCrosse-TX141THBv2-0-102"
		],
		"cns": [
			[
				"mac",
				"LaCrosse-TX141THBv2-0-102"
			]
		],
		"mdl": "LaCrosse-TX141THBv2",
		"name": "LaCrosse-TX141THBv2-0-102",
		"via_device": "lilygo"
	}
}
```
Topic: OpenMQTTlilygo/RTL_433toMQTT/LaCrosse-TX141THBv2/0/102
value:
```
{
	"model": "LaCrosse-TX141THBv2",
	"id": 102,
	"channel": 0,
	"battery_ok": 1,
	"temperature_C": 20.1,
	"humidity": 35,
	"test": "No",
	"mic": "CRC",
	"protocol": "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor",
	"rssi": -71,
	"duration": 463000
}
```

**Expected behavior**
Sensor values to be populated in Home Assistant.

**Environment (please complete the following information):**
* OpenMQTTGateway version 1.7,0

1technophile, 2023-12-18:
Hello,

I think the base topic is missing in your configuration:
`OpenMQTTlilygo/RTL_433toMQTT/LaCrosse-TX141THBv2/0/102`

You should have something like `home/OpenMQTTlilygo/RTL_433toMQTT/LaCrosse-TX141THBv2/0/102`

You can change it in the WebUI Configuration->ConfigureMQTT:
[image]

ZogKarndon, 2023-12-18:
I did have the base topic set, but I didn't have a trailing slash.   Adding the trailing slash seems to have fixed the problem.  Thanks for the pic; it was extremely helpful.

closed 2023-12-18 by ZogKarndon