1technophile/OpenMQTTGateway #1852: Version 1.6 Lacrosse-TX141W not reporting temperature and humidity anymore.

hb9eue, 2024-01-06:
Hi

After Upgrading to 1.6 I don't get temperature and humidity anymore from my 433 MHz LaCrosse-TX141W weather sensor.

Only wind speed and battery are reported.

Here is the message received via MQTT:

`{"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"wind_avg_km_h":0,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-85,"duration":669664}`

And hint appreciated on how to get temperature and humidity back.

-Benoit-

hb9eue, 2024-01-06:
Hmm, also from the OMG console, it looks like two messages are send on the same time, contradicting each other regarding wind speed:

```
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/416073 msg {"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"wind_avg_km_h":5.1,"wind_dir_deg":99,"deg":0,"est":"mic","CRC":"col","sor":-44,"ssi":639000}
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/416073 msg {"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"wind_avg_km_h":0,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-44,"duration":639000}
```

NorthernMan54, 2024-01-06:
The TX 141 device sends out different messages, each containing different portions of the current readings.  And similar to rtl_433, OMG just sends them as received. The client will need to be able to handle this.

My device sends one message with temperature etc, and one with wind speed etc.

Have you tried downgrading to 1.6 to see if the same behavior occurs?

hb9eue, 2024-01-06:
Downgrading to 1.6 fixes the issue.

From the console on 1.6:

```
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/416073 msg {"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"temperature_C":3.8,"humidity":99,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-53,"duration":630748}
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/416073 msg {"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"wind_avg_km_h":0,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-53,"duration":630748}
```

I also looked at the output from 1.7 more closely. I guess the first line is just corrupted information, the string rssi and sensor are truncated.

1technophile, 2024-01-06:
Can you try the development version and let me know if also truncated:
Https://docs.openmqttgateway.com/dev

hb9eue, 2024-01-07:
Devel version looks good. I get all values, MQTT strings don't look corrupted:

```
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/416073 msg {"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"temperature_C":3.3,"humidity":99,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-52,"duration":640000}
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/416073 msg {"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"wind_avg_km_h":0,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-52,"duration":640000}
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/416073 msg {"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"temperature_C":3.2,"humidity":99,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-50,"duration":640992}
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/416073 msg {"model":"LaCrosse-TX141W","id":416073,"channel":0,"battery_ok":1,"wind_avg_km_h":0,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-50,"duration":640992}
```

franki29, 2024-01-10:
Same at my side, with Beta it seems to work . I am using  esp32dev-rtl_433.

franki29, 2024-01-11:
Too early for celebrating, it stopped working after some time

jpmorrison, 2024-02-24:
LaCrosse-TX141THBv2 and TFA-303221 are also gone

Had to downgrade back to 1.6 to fix

franki29, 2024-02-24:
I am running now beta version	"7a3fae" since 15 days without any problems.
In the beginning i had to restart the device several times but now it seems to work.

DigiH, 2024-02-24:
@jpmorrison 

> LaCrosse-TX141THBv2 and TFA-303221 are also gone

Did you also try the development version? I'm getting the LaCrosse-TX141THBv2 fine with it, with it occasionally being mis-recognised as TFA-303221.

Which radio are you using?

jpmorrison, 2024-02-25:
> Did you also try the development version? I'm getting the LaCrosse-TX141THBv2 fine with it, with it occasionally being mis-recognised as TFA-303221.

I haven't yet, I'll try the web upload. When I was on 1.7, trying to change to dev OTA didn't work

> Which radio are you using?

It's a CC1101.  I'm pretty sure it's this module - I have a one with a cheap external antenna, 30 cm/12 dBi (allegedly) and one with the spring antenna.

https://github.com/NorthernMan54/rtl_433_ESP/blob/main/docs/E07-M1101D-TH_Usermanual_EN_v1.30.pdf

DigiH, 2024-02-26:
> It's a CC1101.

Also with a CC1101 here, and using the development branch from about two weeks ago. As I do not recall any issues with the 1.7 release either though it might be something else causing this for you.

StefanHei, 2024-04-13:
I can confirm this problem.
No values for temperature and humidity with 1.7.0 and latest dev-release
lilygo-rtl_433

readings with 1.7.0 (two times):
`{"model":"LaCrosse-TX141W","id":56207,"channel":0,"battery_ok":1,"wind_avg_km_h":2.9,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-46,"duration":642000}
{"model":"LaCrosse-TX141W","id":56207,"channel":0,"battery_ok":1,"wind_avg_km_h":26.9,"wind_dir_deg":27,"deg":0,"est":"mic","CRC":"col","sor":-46,"ssi":642000}
{"model":"LaCrosse-TX141W","id":56207,"channel":0,"battery_ok":1,"wind_avg_km_h":2.5,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-46,"duration":640055}
{"model":"LaCrosse-TX141W","id":56207,"channel":0,"battery_ok":1,"wind_avg_km_h":27,"wind_dir_deg":27,"deg":0,"est":"mic","CRC":"col","sor":-46,"ssi":640055}`

readings with 1.6.0:
{"model":"LaCrosse-TX141W","id":56207,"channel":0,"battery_ok":1,"temperature_C":27.4,"humidity":25,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-46,"duration":638996}

DigiH, 2024-04-13:
> Same at my side, with Beta it seems to work . I am using esp32dev-rtl_433.

> Too early for celebrating, it stopped working after some time

Which frequency have you all set for your rtl_433 gateways, visible and settable in the WebUI?

We recently found that certain stations are worse or better received when tweaking the frequency a bit. This could be especially the case with the development version working fine for the above user but then stopped working after a while.

StefanHei, 2024-04-13:
My frequency is 433.92 MHz.
No function at all with 1.7.0, but perfect with 1.6.0.
Startet today, so i think i need some time to confirm durability with 1.6.0.

DigiH, 2024-04-14:
> My frequency is 433.92 MHz.

Any improvement with the Lacrosse-TX141W reception with 1.7.0 when tweaking the frequency a bit in either direction?

Any possible mis-recognitions of the Lacrosse-TX141W as a TFA 30.3222.02 or TFA 30.3251.10 - although I don't see any such decoders, so likely just still an occasional issue with my Lacrosse-TX141TH-Bv2 vs. TFA 30.3221.02 ;) But then again, this could be exactly the issue, possible mis-recognitions, but going nowhere because of missing TFA 30.3222.02 or TFA 30.3251.10 decoders.

As the actual lacrosse_tx141x.c decoder file hasn't actually changed since April '23 I suppose it could be other factors of radio lib or internal RF changes, which @NorthernMan54 will be more privy to.

Adjusting the frequency in small steps (.01 - .02) up and/or down should be the first step for anyone seeing this with 1.7.0, just to rule out or confirm that this might be the simplest solution to the issue.

closed 2024-07-29 by github-actions[bot]