1technophile/OpenMQTTGateway #1905: ZgatewayBT::launchBTDiscovery theeng properties json deserialization fails with NoMemory on ESP32

ssalonen, 2024-03-03:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**

On 60a571ebb2cebb1f0d176788b82f1fb8ace9218f

`ZgatewayBT::launchBTDiscovery` deserialization fails silenty with DeserializationError::NoMemory on `deserializeJson` [ZgatewayBT.ino:1023](https://github.com/1technophile/OpenMQTTGateway/blob/60a571ebb2cebb1f0d176788b82f1fb8ace9218f/main/ZgatewayBT.ino#L1023) .

This is on ESP32 with `JSON_MSG_BUFFER=768`. Indeed, with Ruuvi tag json properties, even the ArduinoJSON assistant is claiming that 816 bytes is needed at minimum. Indeed, I can avoid crash with `StaticJsonDocument<816> jsonBuffer`

[image]

I noticed this since the program crashes when processing [last entry (`mac`) of the properties json (this is Ruuvi tag)](https://github.com/theengs/decoder/blob/0cdc6bb6f1563f12e3099ba2cf2a78a4dd41a1f1/src/devices/RuuviTag_RAWv2_json.h#L98-L101). Crash stack trace with some debug printing

```
T: Key: mac T: Unit:  T: Name:  T: prop.value()["unit"] null: no T: prop.value()["unit"]: ��?��?��?H%@?p��? T: prop.value()["unit"].as<const char*>() null: yes T: prop.value()["unit"].as<const char*>():  T: prop.key().c_str() null: no T: prop.key().c_str(): mac 

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40091a27  PS      : 0x00060430  A0      : 0x800e3456  A1      : 0x3ffd1590  
A2      : 0x00000000  A3      : 0x3f40459c  A4      : 0x3f403264  A5      : 0x3ffd16e0  
A6      : 0x3ffd1be0  A7      : 0x3f406ad0  A8      : 0x000000a0  A9      : 0x3ffd1560  
A10     : 0x00000000  A11     : 0x3ffd1828  A12     : 0x00000010  A13     : 0x00000021  
A14     : 0x00000001  A15     : 0xff000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40091140  LEND    : 0x4009114b  LCOUNT  : 0x00000000  

Backtrace: 0x40091a24:0x3ffd1590 0x400e3453:0x3ffd15a0 0x400fa0e6:0x3ffd1c00 0x40130711:0x3ffd1c40

  #0  0x40091a24:0x3ffd1590 in strcmp at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strcmp.S:467
  #1  0x400e3453:0x3ffd15a0 in launchBTDiscovery(bool) at main/ZgatewayBT.ino:1091
  #2  0x400fa0e6:0x3ffd1c00 in loop() at main/main.ino:2208
  #3  0x40130711:0x3ffd1c40 in loopTask(void*) at /home/salski/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

```

Note crash happens with `mac` entry, and  `prop.value()["unit"].as<const char*>()` is `NULL`, i.e. not a valid pointer and hence `strcmp` crashes.

This is the printout of the `properties` string:
```
{"properties":{
"hum":{"unit":"%","name":"humidity"},
"tempc":{"unit":"°C","name":"temperature"},
"pres":{"unit":"hPa","name":"pressure"},
"accx":{"unit":"m/s²","name":"acceleration x"},
"accy":{"unit":"m/s²","name":"acceleration y"},
"accz":{"unit":"m/s²","name":"acceleration z"},
"volt":{"unit":"V","name":"voltage"},
"tx":{"unit":"dBm","name":"signal_strength"},
"mov":{"unit":"int","name":"movement counter"},
"seq":{"unit":"int","name":"measurement sequence number"},
"mac":{"unit":"string","name":"MAC address"}}}
```

By some additional debug code, one can confirm that already the `deserializeJson` fails with [`NoMemory`](https://arduinojson.org/v6/api/misc/deserializationerror/#nomemory) ("The [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) is too small; you need to increase its capacity."). Similarly, `jsonBuffer.overflowed()` returns `true`.

**To Reproduce**

BT gateway with Ruuvi tags

**Expected behavior**
BT gateway would not crash with Ruuvi tags

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)

60a571ebb2cebb1f0d176788b82f1fb8ace9218f

* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.

## Possible fix 1: Utilize zero-copy json parsing

Current code uses `std::string` with ArduinoJSON input. This is wasteful are requires more to be allocated with `StaticJsonDocument`. 

We could utilize [zero-copy](https://arduinojson.org/v5/faq/why-is-the-input-modified/#mode-2-zero-copy) mode of ArduinoJSON library to have the parsing pass with the same JSON_MSG_BUFFER. See also tips from the ArduinoJson project: https://arduinojson.org/v6/how-to/reduce-memory-usage/#tip-1-avoid-duplication-if-the-input-is-in-memory

I tried this one and indeed it did not crash at all:

```
char* properties_str; 
{
    std::string properties = decoder.getTheengProperties(p->sensorModel_id);
    properties_str = new char[properties.size() + 1]; // +1 for null terminator
    strcpy(properties_str, properties.c_str()); // Copy the entire string
} 

...
StaticJsonDocument<JSON_MSG_BUFFER> jsonBuffer;
DeserializationError desError = deserializeJson(jsonBuffer, properties_str);
```

The downside with the above is that we temporarily have both `properties_str` and `properties` consuming memory.  Note that this is actually happening with the current code as well since `deserializeJson` [is copying everything with `std::string` input](https://arduinojson.org/v5/faq/why-is-the-input-modified/#mode-1-read-only-input).

I could not figure out a way to avoid this -- one main problem is that theengs decoder API is is allocating string every time:  https://github.com/theengs/decoder/blob/development/src/decoder.cpp#L909C1-L909C4 . The [`const char*`](https://github.com/theengs/decoder/blob/development/src/devices/RuuviTag_RAWv2_json.h#L54) from `_devices` is used with `std::string(const char*)` copy constructor

Instead, theengs decoder API could pass the reference to the static `const char*` (string literal, stored in program data or..?), and that could be copied once to mutable `char*` for JSON parsing purposes.

Or maybe there is a way to get mutable buffer from string, and stop using the string reference? I was not sure about the risks involved (Undefined Behaviour).

## Possible fix 2: increase JSON_MSG_BUFFER

Not sure how viable this is, as the project supports low-mem targets as well. I am also not sure what is the minimum JSON_MSG_BUFFER that would be enough for all properties

The project is not crashing with `StaticJsonDocument<816> jsonBuffer` (minimum for Ruuvi as estimated by [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/#/step3). 

## Possible fix 3: memory-optimized API for Theengs properties json (new-line separated json objects)

I noticed the parsing code at least in ZGatewayBT is processing the properties one-by-one, and it is not even needed to parse all the properties at once. 

With different json format, incremental parsing would be possible, e.g. new line separated json objects

Instead of one-line JSON object
```
"properties":{      "hum":{         "unit":"%",         "name":"humidity"      },      "tempc":{         "unit":"°C",         "name":"temperature"      },      "pres":{         "unit":"hPa",         "name":"pressure"      }, ... }
```

we would have have new line separated json objects:

```
{"hum":{         "unit":"%",         "name":"humidity"      }
{"tempc":{         "unit":"°C",         "name":"temperature"      }
{"pres":{         "unit":"hPa",         "name":"pressure"      }
 ...
``` 
Each line could be read using `deserializeJson(JsonDocument& doc, const char* input, size_t inputSize)`, by pointing the `const char* input` to start of each line.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

ssalonen, 2024-03-03:
@1technophile  I can happily provide a fix for this one, but let me know if there's something else to try out that would cause a crash

1technophile, 2024-03-03:
@ssalonen thanks for the analysis.
I would suggest the fix number 2. Indeed we are planning to remove the support for Arduino boards so increasing the buffer to 816 for ESP32 seems to be the most straightforward approach.

ssalonen, 2024-03-03:
I also note here that ruuvi is having longest properties (json props json string), at 513 bytes. Of course the StaticJsonDocument needs some more for the overhead

This gives a good indication that there are no other tricky devices

ssalonen, 2024-03-03:
I will file a PR bit later.

Will also include error handling

ssalonen, 2024-03-03:
The regression might have happened in recent PR where props have become larger, eg

https://github.com/theengs/decoder/pull/359
Or
https://github.com/theengs/decoder/pull/452

ssalonen, 2024-03-04:
See PR #1906 for the fix.

I also propose we log any deserialization errors, see #1907. In #1907 I did not tweak the implementation any further to avoid access invalid pointers (like what happened here) - I would expect a crash with ESP8266.

PR #1906 (2024-03-04): ESP32 JSON buffer size fixed for RuuviTag_RAWV2
816 bytes is just enough for parsing RuuviTag_RAWV2 json properties.

Resolves #1905

## Description:

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
commit 5432016c (2024-03-04): ESP32 JSON buffer size fixed for RuuviTag_RAWV2 (#1906)

816 bytes is just enough for parsing RuuviTag_RAWV2
json properties.

Resolves #1905
closed 2024-03-04 by 1technophile