1technophile/OpenMQTTGateway #1908: Pin correspondance atk-lora-01

jmguilbert, 2024-03-08:
Hey.
For use the esp32 with lora module, you preconise a module SX12xx.
I have a module atk-lora-01 but the pins no are the same.
I have à pins:
- MDO
- AUX
- RXD
- TXD
- GND
- VCC
Can you help me for correspondance with the pins SCK - GPIO5
MISO - GPIO19
MOSI - GPIO27
SS - GPIO18
RST - GPIO14
DI0 - GPIO26

Thanks.
Jean-Marc.

NorthernMan54, 2024-03-09:
Can you share details for the module ?  Ie a data sheet

jmguilbert, 2024-03-09:
Thank you for your reply.
I don't have much for datasheet.
Here's what I was able to find.
Thank you for your hoped-for help.

[tesi.pdf](https://github.com/1technophile/OpenMQTTGateway/files/14545668/tesi.pdf)
[image]

NorthernMan54, 2024-03-09:
I took a quick look at the data sheet and that board doesn't appear to be a direct replacement for a SX1278, but something else and does not have the appropriate connections to connect.

jmguilbert, 2024-03-09:
Thank for reponse.
Its a sx1278
[image]

NorthernMan54, 2024-03-09:
Googling finds this - https://wiki-power.com/en/Lora通信-基于正点原子ATK-LORA-01模块/

Which indicates that it uses an 'AT' command set, versus the SX1278 which uses a SPI interface and registers

I expect that under the cover you would see a SX1278, and another chip controlling the SX1278.  That is the one that is exposed via the pins.

jmguilbert, 2024-03-09:
I understand that it seems difficult to integrate it. I'll try with another module... Many thanks.

closed 2024-03-09 by NorthernMan54

jmguilbert, 2024-03-13:
@NorthernMan54 
Good evening
You were kind enough to inform me about my ATK-Lora-v3 (433) module and I thank you (1technophile Github).
You explain to me that this lora module uses AT commands and not SPI.
I just bought another lora module "E32 433" from Ebyte and I'm afraid I made the same mistake.
Can you try to explain or give me a link on the difference between the "AT" and "SPI" system.
My goal is to connect one of these modules with ESP32 to interact with OpenMqttGateway but I can't find any resources.
I am a little lost.

NorthernMan54, 2024-03-16:
The Lilygo board is pretty straight forward - no soldering required or wiring required - https://www.aliexpress.com/item/32872078587.html?spm=a2g0o.productlist.main.3.87103pD93pD9kh&algo_pvid=b9ee25ee-de67-4eaf-b559-cf0105eca5ad&algo_exp_id=b9ee25ee-de67-4eaf-b559-cf0105eca5ad-1&pdp_npi=4%40dis%21EUR%2126.75%2120.63%21%21%2128.42%2121.92%21%402101eac917106034660922694ea182%2112000036680562363%21sea%21BE%21799266116%21AC&curPageLogUid=QgXyU7genDYz&utparam-url=scene%3Asearch%7Cquery_from%3A