1technophile/OpenMQTTGateway #1910: Which DIY card do you recommend ?

jmguilbert, 2024-03-10:
To build this example, which card in 433 do you recommend? Can we use a Lilygo Lora 32 v3 for example ? What would be the simplest board without having to use a soldering iron?
Can you provide links to Aliexpress or Amazon?
Big thanks.

NorthernMan54, 2024-03-11:
The one I use is - https://github.com/NorthernMan54/rtl_433_ESP/blob/main/README.md - [LILYGO® LoRa32 V2.1_1.6.1 433Mhz](https://www.lilygo.cc/products/lora3?variant=42476923879605)

jmguilbert, 2024-03-12:
the problem is the very fragile soldering of the pins near the screen cable. did you succeed without burning everything?

jmguilbert, 2024-03-12:
Okay. Thank. I see.
For the sensors (temp, weight etc) you use a DIY products ?

NorthernMan54, 2024-03-12:
I use the lilygo board with Acu-Rite and LaCrosse temperature sensors.  The board receives the transmissions and publishes them.

And in one use case I soldered a BME 280 Sensor to the pins to have a local sensor as well

closed 2024-06-25 by github-actions[bot]