1technophile/OpenMQTTGateway #1916: About the command "String msg ="{xxxx}";

jmguilbert, 2024-03-16:
Hello and thank you for your often quick and helpful responses.
I need to better understand a command you use in the "temperature" example on a DIY sensor.
This is the one I would like to adapt to other sensors:
String msg = "{\"model\":\"ESP32TEMP\",\"id\":\"" + NodeId + "\",\"tempc\":" + String(temp) + "}";

1- I understand that this line sends a message to OMG so that it translates the sensor data into MQTT.
- the "temp" variable comes from float.
But where does “tempc” come from?
Is this a “topic” that OMG can recognize?
If so, what other topics could we use?
For example to send a weight collected by HX711.

2- More generally, could you direct me to a doc so that I can learn how to form a "String msg" by myself without disturbing you.

A big thank you from France.
Jean Marc.

1technophile, 2024-03-18:
> But where does “tempc” come from?

tempc is the key used by OMG to understand that this is a temperature in Celsius

All the properties used in the Theengs decoder could be used, example:
https://github.com/theengs/decoder/blob/121e69989bdfa5831b2f91d181bb838b8e115097/src/devices/XMTZC05HMKG_json.h#L39

In your case it could be:
``` c
String msg = "{"model":"ESP32WGHT","id":"" + NodeId + "","weight":" + String(weight) + "}";
```

jmguilbert, 2024-03-19:
Many thanks.

closed 2024-07-03 by github-actions[bot]