1technophile/OpenMQTTGateway #1930: web flasher not working

deanfourie1, 2024-04-12:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
The web flasher tool is not working. The button just does nothing.

**To Reproduce**
Steps to reproduce the behavior:
Use chrome and attempt to flash a board.

**Expected behavior**
The popup to come up and select a serial port.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

Not installed yet.

**Additional context**
```
Third-party cookie will be blocked. Learn more in the Issues tab.
/manifest.json:1 
        
        
       Failed to load resource: the server responded with a status of 404 ()
/manifest.json:1 Manifest: Line: 1, column: 1, Syntax error.
web-install.html:1 Access to script at 'https://unpkg.com/esp-web-tools@3.4.2/dist/web/start-flash-9b1cffdc.js?module' from origin 'https://docs.openmqttgateway.com' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
unpkg.com/esp-web-tools@3.4.2/dist/web/start-flash-9b1cffdc.js?module:1 
        
        
       Failed to load resource: net::ERR_FAILED
install-button.js?module:1 Uncaught (in promise) TypeError: Failed to fetch dynamically imported module: https://unpkg.com/esp-web-tools@3.4.2/dist/web/start-flash-9b1cffdc.js?module
web-install.html:1 Uncaught (in promise) TypeError: Failed to fetch dynamically imported module: https://unpkg.com/esp-web-tools@3.4.2/dist/web/start-flash-9b1cffdc.js?module
install-button.js?module:1 Uncaught (in promise) TypeError: Failed to fetch dynamically imported module: https://unpkg.com/esp-web-tools@3.4.2/dist/web/start-flash-9b1cffdc.js?module
manifest.json:1 
        
        
       Failed to load resource: the server responded with a status of 404 ()
```
* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2024-04-12:
Hello,

I just tried it and it was working:
[image]

Maybe a transient error

deanfourie1, 2024-04-12:
Are you using latest version of chrome?

1technophile, 2024-04-12:
I'm using Version 123.0.6312.122 (Official Build) (64-bit)
Tested on MacOS also with 122.0.6261.129 and it works

1technophile, 2024-04-12:
Alternatively you could try the development version falsher:
https://docs.openmqttgateway.com/dev/

DigiH, 2024-04-12:
@deanfourie1 

How is this flash erase site working for you?

https://espressif.github.io/esptool-js/

and which OS are you on?

1technophile, 2024-05-17:
Closing as no answer

closed 2024-05-17 by 1technophile

deanfourie1, 2024-05-17:
I got this working, not sure how or why it wasnnt working, just started working