1technophile/OpenMQTTGateway #1940: Mi Band 7: No steps get transfered freely

michi1972-dev, 2024-04-28:
I have the same problem. No steps get transfered freely. Must be a Xiaomi Band 7 firmware issue. Any idea?

_Originally posted by @michi1972-dev in https://github.com/1technophile/OpenMQTTGateway/issues/1642#issuecomment-2081457376_

What can we do? Older Mi Bands working good.

DigiH, 2024-04-28:
Please see my comment in the other closed issue. 
No need to try and open a fresh issue when this has already been discussed in the other thread. 

https://github.com/1technophile/OpenMQTTGateway/issues/1642#issuecomment-2081464791

> What can we do?

If you really want this back you need to pick this up with Xiaomi, and ask them to put the steps information back into the advertising data broadcasts.

https://github.com/1technophile/OpenMQTTGateway/issues/1642#issuecomment-1673089028

closed 2024-04-28 by DigiH

michi1972-dev, 2024-04-28:
OK, I got it. Hoped there may be a chance and was forgotten somehow. Sorry