1technophile/OpenMQTTGateway #1945: not compile in arduino ide

gerashk, 2024-05-11:
Good afternoon. I'm trying to compile it in Arduino according to the instructions. but I get an error
 
"Compilation error: declaration of template parameter 'T' shadows template parameter"

on this line 326 : "void Config_update(JsonObject& data, const char* key, T& var);"

1technophile, 2024-05-11:
Arduino IDE will not be supported in the next release, as announced on v1.7.0
I would suggest to use platformIO instead.

zackfuchtel, 2024-08-10:
Comment line 325 & 326 out. It seems arduino IDE no longer popular also in version 1.7.0...

```
PubSubClient client;

//template <typename T> // Declared here to avoid pre-compilation issue (missing "template" in auto declaration by pio)
//void Config_update(JsonObject& data, const char* key, T& var);
template <typename T>
```

closed 2024-11-23 by github-actions[bot]