1technophile/OpenMQTTGateway #1946: 'to_string' is not a member of 'std' in ZgatewayRF.ino

reimerp, 2024-05-12:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Compile Error:
/workspaces/pio-workspace/main/ZgatewayRF.ino: In function 'void RFtoMQTT()':
/workspaces/pio-workspace/main/ZgatewayRF.ino:134:18: error: 'to_string' is not a member of 'std'
       rawDump += std::to_string(raw[i]);

**To Reproduce**
git checkout v1.7.0
enable +  '-DZgatewayRF="RF"' in environments.ini
pio run -e rfbridge-direct

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
n/a

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.
Processing rfbridge-direct (platform: espressif8266@^2; board: esp8285; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285.html
PLATFORM: Espressif 8266 (2.6.3) > Generic ESP8285 Module
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.20704.0 (2.7.4) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)

```
PlatformIO Core, version 6.1.14
xtensa-lx106-elf-g++ (GCC) 4.8.2

If I remove `-DZgatewayRF="RF"'` project compiles.
I tried to reproduce the error in a very small project, this compiles in esp32 bur not in esp8266
-std is set to gnu++11 in both calls

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

```
#include <Arduino.h>
#include <string>

void setup() {
  std::to_string(0);
}

void loop() {
}

```
```
[env:esp01]
platform = espressif8266
framework = arduino
board = esp01

[env:esp32]
platform = espressif32
framework = arduino
board = esp32dev
```
pio run -e esp01 fails: 'to_string' is not a member of 'std'
pio run -e esp32 works

reimerp, 2024-05-12:
if I change in environments.ini

```
[env:rfbridge-direct]
#platform = espressif8266@^2
with:
platform = ${com.esp8266_platform}

```
the toolchain updates
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285.html
PLATFORM: Espressif 8266 (4.1.0) > Generic ESP8285 Module
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30101.0 (3.1.1) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
```
the code seems to compile, but I got tons on linker errors.

reimerp, 2024-05-13:
I checked out a fresh copy, and also development behaves the way. I replaced the rawdump block in `main/ZgatewayRF.ino` with the one from #a6c0a3b1, which is not relying on std:to_string and build is ok even with on toolchain.

1technophile, 2024-05-13:
If you don't need raw dump, I will comment it.
If you need it, try to use a char array instead.

closed 2024-08-27 by github-actions[bot]