1technophile/OpenMQTTGateway #1948: RTL_433 (esp32 dev+CC1101+Oregon-THGR122N) cyclic reboot and freeze

ASergeyV, 2024-05-19:
Firmware: esp32dev-rtl_433
Version: 1.7.0
Uploaded from web
When receiving a signal from the station Oregon-THGR122N, it restarts and freezes after several reboots. Console output:
rtl_433_ESP(6): data_output {"model":"Oregon-THGR122N","id":251,"channel":1,"battery_ok":1,"temperature_C":27.3,"humidity":14,"protocol":"Oregon Scientific Weather Sensor","rssi":-71,"duration":1754000}
T: isAdupl?
           T: Enqueue JSO
***ERROR*** A stack overflow in task rtl_433_Decoder has been detected.

Backtrace: 0x4008382d:0x3ffe4490 0x4008d7c9:0x3ffe44b0 0x40090ea5:0x3ffe44d0 0x4008f37f:0x3ffe4550 0x4008d8d8:0x3ffe4580 0x4008d888:0x3ffe4364 |<-CORRUPTED

1technophile, 2024-05-19:
Try the development version at https://docs.openmqttgateway.com/dev

ASergeyV, 2024-05-19:
The problem is solved

closed 2024-05-19 by ASergeyV