1technophile/OpenMQTTGateway #1956: Miboxer Lora

domoticaDude, 2024-05-29:
I have an miboxer RF 433MHZ Lora
Model WL 433 can i replace this with openmmqqttgateway?

DigiH, 2024-05-29:
Hi @domoticaDude 

Very unlikely. While one of the RF gateway options might be able to pick up and send to the MiBoxers, it is the proprietary MiBoxer protocol which is unknown to OpenMQTTGateway making it definitely hard or possibly completely impossible to easily set colour, brightness, saturation etc. 

Personally, I am only familiar with the 2.4 GHz RF MiBoxers, for which I use

https://github.com/sidoh/esp8266_milight_hub

Maybe there is a similar gateway project for the 433 Mhz Mi Boxers.

domoticaDude, 2024-06-02:
Thank you! I know the esp8266_milight_hub and that works fine for the milights 2.4 but so far no luck with the lora milights.

closed 2024-09-16 by github-actions[bot]