1technophile/OpenMQTTGateway #1971: Watchdog reset when using nodemcuv2-pilight and receiving rf codes.

Ketturi, 2024-06-16:
ESP8266 crashes hard when using pilight build, and receiver is getting input (eg, by pressing kaku/nexa RF remote)

Serial console output:
```
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: version_tag
N: WiFi ok with manual config credentials
N: Switching to Pilight Receiver: 433.92
N: RF_EMITTER_GPIO: 4 
N: RF_RECEIVER_GPIO: 5 
N: OpenMQTTGateway modules: ["Pilight"]
N: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
N: Connected to broker
N: Send on /RFtoMQTT msg {"active":1}
N: Send on /SYStoMQTT msg {"uptime":5,"version":"version_tag","disc":true,"ohdisc":false,"env":"nodemcuv2-ketturi","freemem":23256,"mqttp":"1883","mqtts":false,"msgprc":0,"msgblck":0,"maxq":0,"eth":false,"rssi":-59,"SSID":"wlanssid","BSSID":"00:11:22:33:44:55","ip":"172.20.1.132","mac":"5C:CF:7F:D9:23:9D","modules":["Pilight"]}

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
N: 
************* WELCOME TO OpenMQTTGateway **************
```
Issue is present with OpenMQTTGateway 1.7.0, and also with current development version.

Environment used:

```
[env:nodemcuv2-ketturi]
platform = espressif8266@^2
board = nodemcuv2
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager8266}
  ${libraries.esppilight}
  ${libraries.esp8266_mdns}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayPilight="Pilight"'
  '-DGateway_Name="OMG_ESP8266_Pilight"'
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="wlanssid"'
  '-Dwifi_password="wlanpw"'
  '-DRF_EMITTER_GPIO=4'
  '-DRF_RECEIVER_GPIO=5'
  '-DENV_NAME="$PIOENV"'
  '-DZmqttDiscovery="HADiscovery"'
  '-DMQTTsetMQTT'
  '-DMQTT_HTTPS_FW_UPDATE'
  '-DMQTT_USER="mqttuser"'
  '-DMQTT_PASS="mqttpass"'
  '-DMQTT_SERVER="172.20.1.100"'
  '-DMQTT_PORT="1883"'
board_build.flash_mode = dout
custom_description = Gateway using ESPilight without the need of CC1101
```

mank4, 2024-06-29:
Hi all, I can confirm the issue. I get exactly the same crash report for the following config:

```
[env:nodemcuv2-pilight-pin45]
platform = espressif8266@^2
board = nodemcuv2
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager8266}
  ${libraries.esppilight}
  ${libraries.esp8266_mdns}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayPilight="Pilight"'
  '-DRF_RECEIVER_GPIO=4'
  '-DRF_EMITTER_GPIO=5'
  '-DGateway_Name="OMG_ESP8266_Pilight"'
board_build.flash_mode = dout
custom_description = Gateway using ESPilight without the need of CC1101
```

There are no crashes with the same config and OpenMQTTGateway 1.6.0.

1technophile, 2024-07-10:
Do you reproduce with the development branch ?

mank4, 2024-10-20:
Thank you for your suggestions. I just tested again, but with the following config as I don't have my nodemcuv2 board with me. I used v1.7.0 and v175, both work fine. I'll test again when I'm back.

```
[env:nodemcuv2-pilight-pin45]
platform = espressif8266@^2
board = esp01_1m
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager8266}
  ${libraries.esppilight}
  ${libraries.esp8266_mdns}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayPilight="Pilight"'
  '-DRF_RECEIVER_GPIO=4'
  '-DRF_EMITTER_GPIO=5'
  '-DGateway_Name="OMG_ESP8266_Pilight"'
  ;'-DLOG_LEVEL=LOG_LEVEL_TRACE'
board_build.flash_mode = dout
custom_description = Gateway using ESPilight without the need of CC1101
```

mank4, 2024-11-02:
I have tested now again with the nodemcuv2 board. I have to add to the last test that I was not aware anymore that the problem only occurs when a signal is received - so that test was completely useless.

I've now put v1.7.0 with the config:
```
[env:nodemcuv2-pilight-pin45]
platform = espressif8266@^2
board = nodemcuv2
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager8266}
  ${libraries.esppilight}
  ${libraries.esp8266_mdns}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayPilight="Pilight"'
  '-DRF_RECEIVER_GPIO=4'
  '-DRF_EMITTER_GPIO=5'
  '-DGateway_Name="OMG_ESP8266_Pilight"'
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
board_build.flash_mode = dout
custom_description = Gateway using ESPilight without the need of CC1101
```
The output is:
```
T: Creating RF PiLight buffer␊
T: Enqueue JSON␊
T: Queue length: 1␊
␍␊
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)␍␊
␍␊
wdt reset␍␊
load 0x4010f000, len 3584, room 16 ␍␊
tail 0␍␊
chksum 0xb0␍␊
csum 0xb0␍␊
v2843a5ac␍␊
~ld␊
N: ␊
************* WELCOME TO OpenMQTTGateway **************␊
N: OpenMQTTGateway Version: version_tag␊
T: mounting FS...␊
T: mounted file system␊
```

However, compilation of the development branch fails with:
```
In file included from .../OpenMQTTGateway/main/main.ino:89:0:
.pio/libdeps/nodemcuv2-pilight-pin45/PicoMQTT/src/PicoMQTT.h:8:2: error: #error PicoMQTT requires ESP8266 board core version >= 3.1
 #error PicoMQTT requires ESP8266 board core version >= 3.1
```
I have not yet figured out how to change that version. Log also says some old version is used (in case this is the version related to the error) although platformio.ini lists a much newer version.
```
PLATFORM: Espressif 8266 (2.6.3) > NodeMCU 1.0 (ESP-12E Module)
```

closed 2025-02-16 by github-actions[bot]