1technophile/OpenMQTTGateway #1995: Add support for Safera Sense / RørosHetta

Takst3in, 2024-08-05:
Hi! Our new ventilation hood for the kitchen stove has an internal unit of "Safera Sense" that is connected via BlueTooth to a PCU unit, controlling the power to the stove. The unit can be directly connected to mobile app (Safera App) via Bluetooth, and while browsing the app for this unit, there are several interesting data that are published from sensors from these units, that would be great to receive into HomeAssistant for Air-quality/etc. 
- Data visible in real-time in the app via BT is; Air-quality index, particulate index, eCO2 ppm, TVOC index, Air temp, Humidity, Alarm level, Heat index, Activity index.

Via OpenMQTTGateway on ESP32, I were able in v1.6.0 to send command below to receive the data-string i suspect contains this info; 
`mosquitto_pub -t home/OMG_ESP32_BLE/commands/MQTTtoBT -m '{
"ble_read_address":"DD:D5:74:67:8E:0D",
"ble_read_service":"0000f00d-1212-efde-1523-785fef13d123",
"ble_read_char":"0000beef-1212-efde-1523-785fef13d123",
"value_type":"HEX",
"immediate": true,
"ttl": 2 }'`

However, this command only returns successfully if the hoos unit is set in Pairing mode prior to sending the command, the command then only gets accepted once. Successive sendings get no data returned. Also the "Successful pairing sound" from the device does not play when returning the data via OMG. 

By this, i suspect that pairing is not successful, there is no PIN required when connecting via the app.

Would be much appreciated if I got the help to aqquire this data via OMG/and successful pairing of the device. 

I have attached a log from nRF Connect (while browsing the app), that shows the connection/data received from the unit. 
[Log 2024-07-30 18_38_24.txt](https://github.com/user-attachments/files/16491206/Log.2024-07-30.18_38_24.txt)

The perfect resolution to this ticket is to get the data decoded my OMG, and published to HA via MQTT for use, any help appreciated!

closed 2024-11-18 by github-actions[bot]

Vispen74, 2025-06-25:
No update on this?

wouterjdb, 2025-07-12:
Interested in the same thing. Though, I think my Røroshetta has Safera cloud as well: thinking that it should be possible to connect to that API and integrate things in HomeAssistant that way. Not very successful with finding any open API though. Any suggestions here?

Takst3in, 2025-07-12:
> Interested in the same thing. Though, I think my Røroshetta has Safera cloud as well: thinking that it should be possible to connect to that API and integrate things in HomeAssistant that way. Not very successful with finding any open API though. Any suggestions here?

Cload only have data from events, not real time sensor data unfortunately.

wouterjdb, 2025-07-12:
Ok, that's a surprising limitation - all hardware is there so it _should_ be possible. But I understand that Bluetooth is the only way then at the moment.

If Røroshetta/Safera read this: highly appreciated if real time (or at least regular data updates) could be supplied through the cloud or local WiFi polling. A local API would be the best, but having an open cloud API that allows for integrating the ventilator with home automation systems would would be a great addition. (Switching ventilator lights on when the rest of the kitchen lights are turned on for example)

Takst3in, 2025-07-12:
> Ok, that's a surprising limitation - all hardware is there so it _should_ be possible. But I understand that Bluetooth is the only way then at the moment.
>
> If Røroshetta/Safera read this: highly [...]

Yes, very limited unfortunately..

Safera brags about 'ready for all protocols' incliding ZigBee on their website, however its not supported..

I have two tickets open at Safera Support:
- When wull ZigBee be ready
- Documentation for BLE protocol/data

Unfortunately both tickets have been open for over a year, with NO! answer/feedback other that 'i will check with a collegue'.. I also congratulated them with the 1 year birthday for the ticket some time ago..

magicus, 2025-12-26:
That's too bad. :( I am also trying to get access to the data from my Safera, and this is just about the only reference my googling turned up.

Do you have any links to these tickets, or are they just private to you?

I'm getting a bit interested in trying to reverse engineer the bluetooth protocol myself, if no-one have done that. Hopefully they have not put too much effort into making the protocol complicated, so it might be just to sniff the traffic for the app, see how it displays the data, and then analyzing it and trying to figure out what's what in the data. But then again, I haven't worked so much with low-level bluetooth access, so I only have a vague idea about GATT and whatnot.