1technophile/OpenMQTTGateway #1996: Add ability to change discovery prefix on the fly either via WebGUI or mosquitto_sub

puterboy, 2024-08-05:
**Is your feature request related to a problem? Please describe.**
I am using OMG_lilygo_rtl_433_esp to capture 433 messages from my various sensors around my house and then post them to my Mosquitto MQTT broker for use in HomeAssistant.

- On the one hand, I need to **shut off** OpenMQTT Gateway "Auto Discovery" because otherwise HA dutifully recognizes and inserts new devices and sensors for the dozens of devices that my neighbors have (which then get further duplicated whenever they or even I change batteries in various devices).

- On the other hand, when I add a new device myself, I want to generate a discovery message(s) for that device alone so that HA will recognize that sensor -- but I don't want to suck in my neighbors devices at the same time

**Describe the solution you'd like**
I would like to have the discovery prefix be changeable on-the-fly without having to recompile and re-flash firmware.
This could done either with the Web GUI (similar to how one can set the `MQTT Base Topic` from the `Configure MQTT` menu or by sending a "System Command" via mosquitto_pub as in https://docs.openmqttgateway.com/use/gateway.htm.

So one could have:
```
mosquitto_pub -t "home/OpenMQTTGateway/commands/MQTTtoSYS/config" -m '{  "mqtt_discovery_prefix": "homeassistant-staging" }
```
which would be analogous to how one sets the MQTT_topic:
```
mosquitto_pub -t "home/OpenMQTTGateway/commands/MQTTtoSYS/config" -m '{  "mqtt_topic": "topic/" }
```

**Describe alternatives you've considered**
I have considered the following 3 workarounds:

1. Recompile and reflash the firmware for my Lilygo device with the Discovery Prefix changed to something other than the default discovery prefix used by HA (which is "homeassistant") and then manually cut and paste my newly discovered device from the new discovery prefix to the HA assistant one (or use mosquitto_sub/mosquitto_pub to do the same)

2. Change either temporarily or permanently the default discovery prefix for HA away from the default discovery prefix for OpenMQTTGateway (which is "homeassistant"). Then delete any unwanted neighbor discovery messages and turn off OpenMQTT Gateway auto-discovery before changing the HA discovery prefix back to its default

3. Turn off HA or shut off MQTT integration polling while I listen for discovery messages from OpenMQTT Gateway and then delete any unwanted newly discovered neighbor devices and turn off auto-discovery before turning back on the HA MQTT integration

However, changing the Discovery prefix on the fly within OpenMQTTGateway seems the easiest and cleanest solution that does the least "harm" since I don't disrupt my HA setup and I can easily change the OpenMQTTGateway prefix back to its default when I am done adding any new devices.

puterboy, 2024-08-15:
It would seem adding this would require only three almost trivial changes to the code base:
(Note: my formatting below is pseudo-patch like)

**[main/config_mqttDiscovery.h](https://github.com/1technophile/OpenMQTTGateway/blob/1c942a06ccbe0603999b51b09571fc2965c6d504/main/config_mqttDiscovery.h#L115)**
```
  #ifndef discovery_Topic
  #  define discovery_Topic "homeassistant"
  #endif
...
+ char discovery_topic[parameters_size + 1] = discovery_Topic;
```

**[OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway/tree/1c942a06ccbe0603999b51b09571fc2965c6d504)/[main](https://github.com/1technophile/OpenMQTTGateway/tree/1c942a06ccbe0603999b51b09571fc2965c6d504/main)
/main.ino**
```
 if ((SYSdata.containsKey("mqtt_topic") && SYSdata["mqtt_topic"].is<char*>()) ||
        (SYSdata.containsKey("gateway_name") && SYSdata["gateway_name"].is<char*>()) ||
+     (SYSdata.containsKey("discovery_topic") && SYSdata["discovery_topic"].is<char*>()) ||
        (SYSdata.containsKey("gw_pass") && SYSdata["gw_pass"].is<char*>())) {
      if (SYSdata.containsKey("mqtt_topic")) {
        strncpy(mqtt_topic, SYSdata["mqtt_topic"], parameters_size);
      }
      if (SYSdata.containsKey("gateway_name")) {
        strncpy(gateway_name, SYSdata["gateway_name"], parameters_size);
      }
+   if (SYSdata.containsKey("gateway_name")) {
+     strncpy(discovery_topic, SYSdata["discovery_topic"], parameters_size);
+   }
      if (SYSdata.containsKey("gw_pass")) {
        strncpy(ota_pass, SYSdata["gw_pass"], parameters_size);
        restartESP = true;
      }
```

**[ain/main.ino](https://github.com/1technophile/OpenMQTTGateway/blob/1c942a06ccbe0603999b51b09571fc2965c6d504/main/main.ino#L343)**
```
- const bool is_autodiscovery_subscription = (pattern == "#") || (pattern.startsWith(String(discovery_Topic) + "/"));
+ const bool is_autodiscovery_subscription = (pattern == "#") || (pattern.startsWith(String(discovery_topic) + "/"));
```
Couldn't be more trivial :)

If you are willing to add this feature, I will test it and create a proper PR.
Let me know...

1technophile, 2024-08-15:
Feel free to push a PR, also would be interesting to have the capability into the WebUI

puterboy, 2024-08-15:
Trying to understand the WebUI code but if I am reading the code correctly, it seems like it will require only minor additions to `config_WebContent.h` to extend the web page and to `ZWebUI.ino` to handle the page.

Though I am a little confused about WEBtoSYS and what that does vs. just directly setting the variables based on the data coming back from the web form.

puterboy, 2024-08-15:
**https://github.com/1technophile/OpenMQTTGateway/blob/development/main/config_WebContent.h**
```
- // mqtt server (mh), mqtt port (ml), mqtt username (mu), mqtt password (mp), secure connection (sc), server certificate (msc), topic (mt)
+ // mqtt server (mh), mqtt port (ml), mqtt username (mu), mqtt password (mp), secure connection (sc), server certificate (msc), topic (mt), discovery topic (dt)
...
- const char config_mqtt_body[] = body_header "<fieldset class=\"set1\"><legend><span><b>MQTT Parameters</b></span></legend><form method='get' action='mq'><p><b>MQTT Server</b><br><input id='mh' placeholder=" MQTT_SERVER " value='%s'></p><p><b>MQTT Port</b><br><input id='ml' placeholder=" MQTT_PORT " value='%s'></p><p><b>MQTT Username</b><br><input id='mu' placeholder=" MQTT_USER " value='%s'></p><p><label><b>MQTT Password</b></label><br><input id='mp' type='password' placeholder=\"Password\" ></p><p><b>MQTT Secure Connection</b><br><input id='sc' type='checkbox' %s></p><p><b>Gateway Name</b><br><input id='h' placeholder=" Gateway_Name " value=\"%s\"></p><p><b>MQTT Base Topic</b><br><input id='mt' placeholder='' value='%s'></p><br><button name='save' type='submit' class='button bgrn'>Save</button></form></fieldset>" body_footer_config_menu;

+ const char config_mqtt_body[] = body_header "<fieldset class=\"set1\"><legend><span><b>MQTT Parameters</b></span></legend><form method='get' action='mq'><p><b>MQTT Server</b><br><input id='mh' placeholder=" MQTT_SERVER " value='%s'></p><p><b>MQTT Port</b><br><input id='ml' placeholder=" MQTT_PORT " value='%s'></p><p><b>MQTT Username</b><br><input id='mu' placeholder=" MQTT_USER " value='%s'></p><p><label><b>MQTT Password</b></label><br><input id='mp' type='password' placeholder=\"Password\" ></p><p><b>MQTT Secure Connection</b><br><input id='sc' type='checkbox' %s></p><p><b>Gateway Name</b><br><input id='h' placeholder=" Gateway_Name " value=\"%s\"></p><p><b>MQTT Base Topic</b><br><input id='mt' placeholder='' value='%s'></p><b>MQTT Discovery Topic</b><br><input id='dt' placeholder='' value='%s'></p><br><button name='save' type='submit' class='button bgrn'>Save</button></form></fieldset>" body_footer_config_menu;
```
**https://github.com/1technophile/OpenMQTTGateway/blob/development/main/ZwebUI.ino**
```
 if (server.hasArg("mt")) {
        WEBtoSYS["mqtt_topic"] = server.arg("mt");
        if (strncmp(mqtt_topic, server.arg("mt").c_str(), parameters_size)) {
          update = true;
        }
      }

1technophile, 2024-08-15:
> Trying to understand the WebUI code but if I am reading the code correctly, it seems like it will require only minor additions to config_WebContent.h to extend the web page and to ZWebUI.ino to handle the page.

Correct

> Though I am a little confused about WEBtoSYS and what that does vs. just directly setting the variables based on the data coming back from the web form.

WEBtoSYS is a way to pass the parameter to MQTTtoSYS

1technophile, 2024-08-15:
I am not sure we should enable to change the discovery message during onboarding, I think a new parameter may confuse the users and generate support.

puterboy, 2024-08-15:
OK - does that mean not adding the WiFi manager codelines? (if I am understanding that correctly wifiManager is used to create a temporary AP to log into, right? -- if I am misunderstanding, please point me to the code that is run during onboarding)

BTW, I made a couple of corrections to the above pseudo-patches

puterboy, 2024-08-15:
BTW, I think `discovery_prefix` might be a better name for the new variable version of the declaration `discovery_Topic`
I say this because you already use the variable name `discovery_topic` for the name of the discovery (sub) topic that sits under what is now the fixed `discovery_Topic`.

I could of course just set: `discovery_prefix = discovery_Topic` but ideally, one would rename`discovery_Topic` to `discovery_Prefix` to be consistent between naming of the declaration and its associated variable  (Note that even now it is confusing that the declaration  `discovery_Topic` refers to the base parent while the variable `discovery_topic` refers to the child subtopic)

Note that Home Assistant FWIW uses the term `discovery_prefix`

In order not to break existing configs, perhaps `discovery_Topic` could be deprecated in favor of `discovery_Prefix` for the declaration, but an `#ifdef` could be added to pull in `discovery_Topic` if it exists in someone's (legacy) config.

Let me know what you think

1technophile, 2024-08-15:
> OK - does that mean not adding the WiFi manager codelines? 

Yes, Indeed. The onboarding can be done through a web portal. Most of the time, when used on a mobile phone, adding a field to the limited screen may increase the risk of misconfiguration by non-initiated users.

The parameters that are not into WiFi Manager but saved are into `SYSConfig_s`, it could be added there.

> BTW, I think discovery_prefix might be a better name for the new variable version of the declaration discovery_Topic
I say this because you already use the variable name discovery_topic for the name of the discovery (sub) topic that sits under what is now the fixed discovery_Topic.

It makes sense

> In order not to break existing configs, perhaps discovery_Topic could be deprecated in favor of discovery_Prefix for the declaration, but an #ifdef could be added to pull in discovery_Topic if it exists in someone's (legacy) config.

Agreed

puterboy, 2024-08-15:
One (hopefully) final question...
I am a bit confused about how & when changes are stored to flash or other non-volatile storage.
It seems like for certain variables (e.g, mqtt_topic, gatewa_name, gw_pass, mqtt_topic, etc.) saving is always done when variables are changed via the 'saveConfig' routine which saves a JSON'ized string of the current variable settings.

On the other hand, other variables (e.g., discovery, ohdiscovery, XtoMQTT, rgbbrightness), use a separate 'preferences' way to save the values to NVS and that only happens when the 'save' flag is set in the mqtt payload.

Why are these changes treated differently?
Am I missing something?

1technophile, 2024-08-15:
The first approach is due to the use of WiFi Manager for onboarding. After that, parameters were added that did not require adjustment through the onboarding process; they were added through the NVS/preferences approach.

puterboy, 2024-08-26:
Merged with Development branch

puterboy, 2024-08-27:
Just found a small but critical issue in my patch.
In the new stanza in `MQTTtoSYS` where `discovery_prefix` is parsed, I added the line:
```
restartESP = true
```
which I inserted so that discovery topics will be generated fresh under the new `discovery_prefix`
HOWEVER, based on how `handleMQ` in `ZwebuUI.ino` is coded, this has the side-effect of triggering a reboot for **every** change on the MQTT Configuration page which in turn creates a problem for the MQTT login variables (e.g., `mqtt_server`, `mqtt_port`, `mqtt_user`, `mqtt_pass`) since they are only saved as part of a call to `setupMQTT()` triggered by setting `mqttSetupPending` which then never happens as it is pre-empted by the reboot which wipes out any changes to those unsaved variables.

Now I realized that one really doesn't even need to reboot when `discovery_prefix` is changed.
All one deeds to do is call `setupMQTT()` (which happens anyway) with `connectedOnce = false` to mimic a first connection.
Unfortunately, that variable is a static variable internal to `handle_autodiscovery`.

Would you be OK making this a global variable within `main.ino`?
Then I could just replace the line `restartESP = true` in the `discovery_prefix` stanza with the line `connectedOnce = false`.
It's probably not a bad global to have access to anyway as its usage can more generally be extended in the future to simulate starting MQTT again for other purposes without having to reboot.

puterboy, 2024-08-27:
**More generally,** it's not clear to me why any of the other variables require a reboot.
- The WiFi-related variables would just require a reconnect to WiFi  followed by a reconnect to MQTT if not already trigerred automatically
- The Gateway name and password similarly shouldn't require a reboot. The password shouldn't require anything (now it requires a reboot) and the name should just trigger an MQTT connect, potentially with a `connectedOnce = false` since many of the MQTT topic names include the gateway name somewehere (right now changing the name just triggers a reconnect which is probably not sufficient).

puterboy, 2024-08-27:
BTW, the reason I suggest ultimately a rewrite would be nice is that the code in `MQTTtoSYS`  seems a bit hairy and inconsistent at times -- presumably it was built up over time.

For example, some variables directly and explicitly trigger a `saveConfig` and/or `ESPrestart` while others do so only very indirectly via a combination of calling `setupMQTT` and connecting to MQTT which triggers a save and reboot as a side effect of the fact that other variables that happen to be on the same page are stored in the `cnt_parameters_array`.

Also, as a result of the above, **some variables actually trigger `saveConfig` twice resulting in duplicated saving and wear-and-tear on the flash** -- first via an explicit call to `saveConfig` and second via setting the `mqttSetupPending` which in turn calls a subsequent `setupMQTT` and an eventual further call to `saveConfig` (as described above)

puterboy, 2024-08-29:
Fixed as per above PRs

closed 2024-08-29 by puterboy