1technophile/OpenMQTTGateway #1998: jeedom - any command to switchbot X1

anthopocket, 2024-08-09:
hi,

i have a problém with the switchbot X1. I have any command to mqttdiscovery

[image]
[image]

in the mqtt broker, the log is

`[2024-08-09 21:23:31] DEBUG  : new message:{"home":{"OMG_GARAGE":{"OMG_GARAGE":{"BTtoMQTT":{"E6B316D83CE6":{"id":"E6:B3:16:D8:3C:E6","rssi":-77,"brand":"SwitchBot","model":"Bot","model_id":"X1","type":"ACTR","mode":"onestate","state":"on","batt":100}`

the gateway is
[image]

Thanks you

anthopocket, 2024-08-09:
communty jeedom

https://community.jeedom.com/t/ne-reconnait-pas-le-switchbot-alors-connu-domg/129100

1technophile, 2024-08-09:
What is the bot firmware version?

anthopocket, 2024-08-10:
Firmware : 4.9

closed 2024-08-13 by 1technophile