1technophile/OpenMQTTGateway #2022: Add ability to log more data to MQTT

puterboy, 2024-08-25:
Not sure if this is a feature request or a bug :)
The MQTT topic `LOGtoMQTT` seems to only log a single entry at startup/restart of the device, namely something like:
```
  "reason": 6,
  "uptime": 2490
```

Changing the Log Level in the GUI has no effect.
I (perhaps naively) expected changing the log level in the GUI to allow for more verbose logging to MQTT.

Would it make sense to allow for more verbose logging to MQTT so that you aren't restricted to logging via the serial interface which is a PITA when your receiver is distant from your PC work area.

1technophile, 2024-08-25:
we used to have an MQTT logger a few years ago but the quantity of messages were generating instabilities.
It may not be the case anymore with the central queue implementation.

puterboy, 2024-08-30:
Perhaps instabilities were caused by the memory corruption bug in jsonQueue that is now fixed?

1technophile, 2024-08-31:
It may be.
I'm currently improving the mqtt msg publish frequency. Still encountering issues, this work could be a prereq to an mqtt log publication.

puterboy, 2024-11-30:
Would still be good to get a way to get more verbose logging via MQTT.
For example, this would be important for some of my decoder debugging since the signals that are causing my crash are picked up only by my downstairs installation and not near my computer where I can connect to the serial port...

puterboy, 2025-03-02:
Please keep open as this is valuable for debugging.

puterboy, 2025-06-01:
Please keep open

puterboy, 2026-09-01:
Please keep open