1technophile/OpenMQTTGateway #2023: Display delay could potentially lead to queue overflow

puterboy, 2024-08-25:
Devices like the LILYGO LoRa32 display the MQTT message on their internal OLED display.
The corresponding code in `ZdisplaySSD1306` hard-codes a 2 second delay to allow the message to display.

I believe the current MQTT queue length is set to 18, meaning that when the queue is full it will take at least 36 seconds to empty it due to the display delays. So if you need to send on average more than 1 MQTT every 2 seconds, then the queue will eventually fill and overflow, no matter how large it is.
This is not a crazy edge case if you (and your neighbors) have lots of random rtl_433 devices (and at least in my neighborhood, there are lots of them).

It might make sense to have this delay be adaptive as the MQTT queue fills up.
Say:
If queue less than half full, full 2 seconds
If queue more than half but less than 3/4 say 1 second
If queue more than 3/4 full, then skip display

(or any other similar schedule)

1technophile, 2024-08-25:
We made the choice when adding the WebUI and display to keep the messages during a bit long for visibility, without it, it was difficult to read the messages. They were going fast.
@NorthernMan54 may have an idea for this one

puterboy, 2024-08-25:
That's why I proposed my solution as a compromise. Keep the visibility long but shorten it when the queue is filling because then you risk skipping (i.e. losing) messages entirely which is certainly worse case.
With my solution plus/minus tuning on the percentages you get the best of both worlds.  I

I will code it and try it and share as PR.

1technophile, 2024-08-25:
An alternative could be to optimize the screen area usage, but not sure if there is too much room for this

puterboy, 2024-08-26:
Would it be possible to put the display in a separate thread? (again I don't know much about multi-threading).
I say this because in general one presumably wants updated MQTT messages to be published as near real-time as possible whereas now if you have multiple sensors sending data, the delay could be several tens of seconds.

If it can't be put in a separate thread perhaps it could be added to a queue that is checked once per main loop to see if 2 seconds have elapsed since the last message.

For example if I am monitoring voltage and current spikes, I may want to see it now.
If I have an alarm, I want it now, not in some arbitrary delay amount of time.

In most if not all cases, presumably the display is a secondary feature and it doesn't seem to make sense to have it time limit the MQTT reporting.

NorthernMan54, 2024-08-26:
@puterboy Take a look at the code behind the webui and ssd1306 web module.  They are implemented this way.

When we implemented the json message display for WebUI and SSD1306, we had included logic to throw away messages once the queue is full.  Thought was that in a high volume situation, like your example, the user wouldn't be using the screen anyway.  Personally I think of it as a parlour trick, looks interesting, makes good YouTube videos, but in reality is not a practical method of sharing information.

puterboy, 2024-08-26:
Ahhh. Great. So the display runs as a loop in a separate thread defined in `ZdisplaySSD1306.ino`  rather than causing everything to wait -- that's perfect! (assuming I am understanding this correctly)

I was though and remain confused by the call in  `ZgatewayRTL_433.ino`  to`pubOled` which I only see mentioned as a blank declaration in `User_config.h`
```
#define pubOled(...)        // display the published message onto the OLED display
```
But now I see that the actual display is being done by `MQTTtoSSD1306()` which is called from `receivingMQTT()` in `main.ino`.
If I am now understanding this correctly, what is the purpose of the do-nothing `pubOled` call?

NorthernMan54, 2024-08-26:
pubOled is a remnant of the original approach, that has been replaced with the messages going thru the webui, then to ssd1306.

puterboy, 2024-08-29:
Should I submit a PR to delete `pubOled` or do you want to keep it there?

NorthernMan54, 2024-08-29:
Your okay to remove it

NorthernMan54, 2024-08-29:
Tks for your recent efforts, much appreciated

puterboy, 2024-08-29:
Your very welcome - least I could do given all the work you guys have put in.

closed 2024-08-29 by puterboy