1technophile/OpenMQTTGateway #2042: Web UI lacks configuration option for Home Assistant discovery

Forge36, 2024-09-01:
**Is your feature request related to a problem? Please describe.**
If the gateway becomes lost to Home Assistant and discovery is disabled it's not possible to re-enable via the web UI.

This makes re-enabling difficult as it requires knowing how to re-enable it via MQTT (or the Web UI console?)

**Describe the solution you'd like**
The discovery state is presented in "Information"
I'd like a visible toggle in "Configuration" (likely under "gateway")
to allow re-enabling discovery.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Per: https://community.openmqttgateway.com/t/stuggling-again-with-home-assistant-mqtt-discovery/2992/16

Send two command to `<device>/commands/MQTTtoSYS/config`

1. `{"disc":true,"save":true}` to re-enable discovery
2. `{"cmd":"restart"}`

 

**Additional context**
Consider adding these options as well
[image]

puterboy, 2024-09-09:
I think that is is a good suggestion and would be quite easy to add to the MQTT page.
In fact, personally, I think it makes more sense here than in HA as you should be able to turn it on or off regardless of what dashboards you have set up in HA (or even if you are using HA).

TBH, when I first started using OpenMQTTGateway and the HA MQTT Integration, I was confused that to turn on auto-discovery and automatically populate new devices, you need to make sure it is turned on both in OMG and in the HA MQTT Integration (to stop auto-creation of new devices you just need to disable one of them).

This was not at all obvious to me as a newbie.

Having an option to turn/on OMG auto-discovery in the OMG GUI would make a lot of sense.

1technophile, 2024-09-09:
It makes sense to add it. I would like to find a way to reflect the parameters for each module comprehensively.
I think all the parameters could be exposed into the WebUI mainly. Instead of maintaining the web page for each parameter set manualy, we could try to generate them depending on the parameters in SYSConfig, RFConfig, and BTConfig.

closed 2024-12-24 by github-actions[bot]