1technophile/OpenMQTTGateway #2044: BLE data for aranet4 sensor not getting decoded

ys333193, 2024-09-04:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I don't see any data published except for id, name ans rssi for brand new Aranet4 sensor, firmware v 1.4.19, "Smart home Integration" option enabled and bluetooth dip switch set to on.  At the same time, Home Assistant integration picks up the sensor data just fine.
Aranet4 listed as supported.

**To Reproduce**
Steps to reproduce the behavior:
1. Brind Aranet4 sensor in proximity of the Gateway
2. Observe data published to MQTT
3. No temperature, humidity, pressure, CO2 or battery data is published, only sensor id, name and rssi.

**Expected behavior**
Temperature, humidity, pressure, CO2 and battery data from the sensor are published.

**Screenshots**
Sample MQTT payload (ID partially masked):
{
  "id": "60:C0:BF:XX:X:XX",
  "name": "Aranet4 224E5",
  "rssi": -80
}

**Environment (please complete the following information):**
* OpenMQTTGateway version used: v1.7.0, release
* Env: lilygo-ble, uC: ttgo-lora32-v21, uploaded with web uploader

**Additional context**
Add any other context about the problem here.

DigiH, 2024-09-05:
Hi @ys333193 

Could you set [Advertisement and Advanced Data](https://docs.openmqttgateway.com/use/ble.html#advanced-advertisement-and-advanced-data-default-false-available-with-ha-discovery) to **true**, and then post some more sample MQTT messages, but this time along with the undecoded advertising data, which let's us see why your brand new Aranet4 with firmware 1.4.19 is not being correctly recognised and decoded.

ys333193, 2024-09-05:
Actually, I've updated to the latest Dev version (https://docs.openmqttgateway.com/**dev**) and sensor data is decoding just fine.

DigiH, 2024-09-05:
Ok, then it was just the Aranet4 decoder not already being present in the last 1.7.0 release. 😃 

Closing

closed 2024-09-05 by DigiH