1technophile/OpenMQTTGateway #2064: BME280 not working (Bosch BME280/BMP280 failed: 0x0) - Heltec v2 running OMG heltec-rtl_433-fsk

mcmlxxvii, 2024-09-24:
I'm facing the exact same issue with a BME280 and Heltec v2 (heltec-rtl_433-fsk development). I'm using this one: https://www.amazon.ca/dp/B07KYJNFMD 

That sensor is an I2C only version that works both in 3.3v and 5v, as it has a built-in voltage regulator. Mine is connected to 3.3v though.

I tested 3 different sensors (using both 0x76 and 0x77 addresses) with my Heltec v2 board, using the recommended pinout in OMG's documentation: 2 of the [aforementioned model](https://www.amazon.ca/dp/B07KYJNFMD) and 1 of a [similar model](https://www.amazon.ca/dp/B0CXF89ZYQ) which has 6 pins and is a hybrid I2C/SPI version. All of them show the same behavior: _Bosch BME280/BMP280 failed: 0x0_ and weird readings, preceded by _[E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263_ error message in the serial console.

I tested all those sensors with an ESP-WROOM-32 board using the exact same wiring, voltage and pinout. For that, I leveraged an I2C Scanner Sketch to make sure of the sensor's address and a BME280 Testing Sketch using both SparkFun (2.0.9) and Adafruit (2.2.4) libraries in Arduino IDE.

Everything worked as expected using the ESP-WROOM-32. However, when trying the exact same sensors with a Heltec v2 running OMG heltec-rtl_433-fsk compiled via Gitpod.io, they don't work.

You can see some console logs illustrating the issue below:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
N:
************* WELCOME TO OpenMQTTGateway **************
N: SYS config not found
N: SSD1306 config initialised
N: Saved SSD1306 config loaded
N: Setup SSD1306 Display end
N: OpenMQTTGateway Version: 20240918-01
N: OTA server cert hash: 02118cbaf5cc02748398d93ecdb0ef2a444c37358a617bfb2206dfe                                                 351dc7cf2
N: Config loaded from flash
N: Attempting Wifi connection with saved AP: 0
N: No WebUI config to load
N: OpenMQTTGateway URL: http://[redacted by me]/
N: ZwebUI setup done
N: RF Config not found using default
N: Enable RTL_433 Receiver: 915.00Mhz
N: Setup BME280/BMP280 on address: 0x76
N: Setup BME280/BMP280 on SDA pin: 0x15 ----> Added by me for debugging
N: Setup BME280/BMP280 on SCL pin: 0x16 ----> Added by me for debugging
[  8943][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
N: Bosch BME280/BMP280 failed: 0x0
N: ZgatewayRTL_433 setup done
N: Offline disabled
N: OpenMQTTGateway modules: ["HELTEC_SSD1306","WebUI","BME280","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
N: Reconfiguring MQTT client...
N: Connected to broker
N: [ OMG->MQTT ] topic: home/OMG_heltec_rtl_433_ESP_FSK/LWT msg: online
```

```
[462430][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[463432][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[464433][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[465434][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[466435][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[467436][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
N: [ OMG->MQTT ] topic: home/OMG_heltec_rtl_433_ESP_FSK/CLIMAtoMQTT/bme msg: {"tempc":-3.4,"tempf":25.88,"hum":0,"pa":0,"altim":44330.77,"altift":145442.2}
```
Hope the above information helps! I'm having a hard time trying to figure out what's going on by myself - This is driving me nuts. ;)

Other than that, OMG is working amazingly well in that board, reading a couple of WH51 915Mhz sensors in my yard and reporting back to HASS.

Thanks!

[image]
[image]
[image]
[image]
[image]

NorthernMan54, 2024-09-24:
So I went back to look at my devices, as I use the BME280 with the LilyGo board and have the Heltec board in my test setup.

This is the web console from my LilyGo board
[image]

```
N: OpenMQTTGateway URL: http://192.168.1.101/
N: ZwebUI setup done
N: RF Config not found using default
N: Enable RTL_433 Receiver: 433.92Mhz
N: Setup BME280/BMP280 on address: 0x76
N: Bosch BME280 successfully initialized: 0x60
N: ZgatewayRTL_433 setup done 
N: Offline disabled
N: OpenMQTTGateway modules: ["LILYGO_OLED","WebUI","BME280","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
N: Reconfiguring MQTT client...
N: Connected to broker
N: [ OMG->MQTT ] topic: home/OMG_4C7525A77544/LWT msg: online 
N: [ OMG->MQTT ] topic: home/OMG_4C7525A77544/RFtoMQTT msg: {"active":3,"frequency":433.92,"rssithreshold":-82,"rssi":-81,"avgrssi":0,"count":0,"ookthreshold":15} 
N: Update check, free heap: 105948N: Using memory cert
N: Update check done, free heap: 59564N: [ OMG->MQTT ] topic: home/OMG_4C7525A77544/SYStoMQTT msg: {"uptime":15,"version":"South-LilyGo-v1.7.0-152-g0d338cc5[development]","env":"South-LilyGo","freemem":109368,"mqttp":"1883","mqtts":false,"mqttv":false,"msgprc":0,"msgblck":0,"msgrcv":0,"maxq":0,"cnt_index":0,"minmem":95740,"tempc":48.89,"freestck":4576,"powermode":-1,"eth":false,"rssi":-54,"SSID":"The_Beach","BSSID":"90:72:40:18:A7:BE","ip":"192.168.1.101","mac":"4C:75:25:A7:75:44","modules":["LILYGO_OLED","WebUI","BME280","rtl_433"]} 
N: [ OMG->MQTT ] topic: home/OMG_4C7525A77544/SSD1306toMQTT msg: {"onstate":true,"brightness":50,"display-flip":true,"idlelogo":true,"log-oled":false,"json-oled":true} 
N: [ OMG->MQTT ] topic: home/OMG_4C7525A77544/WebUItoMQTT msg: {"displayMetric":true,"webUISecure":false,"displayQueue":0} 
N: [ OMG->MQTT ] topic: home/OMG_4C7525A77544/RLStoMQTT msg: {"latest_version":"v1.7.0","title":"OpenMQTTGateway","release_url":"https://github.com/1technophile/OpenMQTTGateway/releases/tag/v1.7.0","installed_version":"South-LilyGo-v1.7.0-152-g0d338cc5[development]","entity_picture":"https://github.com/1technophile/OpenMQTTGateway/raw/development/docs/img/Openmqttgateway_logo_mini_margins.png","release_summary":""} 

N: [ OMG->MQTT ] topic: home/OMG_4C7525A77544/CLIMAtoMQTT/bme msg: {"tempc":23.55,"tempf":74.39,"hum":60.42188,"pa":99431.5,"altim":158.8248,"altift":521.0789} 
```

This is from my Heltec 915 Mhz board, that I'm using for FSK testing

[image]

```
N: OpenMQTTGateway URL: http://192.168.1.96/
N: ZwebUI setup done
N: Config frequency changed: false
N: RF Receive mhz: 915.00
N: RF receiver active: 3
N: Config active unchanged: false
N: Enable RTL_433 Receiver: 915.00Mhz
N: RF Config loaded
N: Setup BME280/BMP280 on address: 0x76
N: Bosch BME280/BMP280 failed: 0x0
N: ZgatewayRTL_433 setup done 
N: Offline disabled
N: OpenMQTTGateway modules: ["HELTEC_OLED","WebUI","BME280","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
N: Reconfiguring MQTT client...
N: Connected to broker
```

And the Sensor ( I used the same physical sensor when testing, to remove a bad sensor as an issue )

[image]

After spending some time this AM on this, I can only think that there is something off with the Heltec board and I2C.  As the exact same code and BME280 worked on the LilyGo and not on the Heltec.  I also tried using different pins for I2C, and had the same result.

mcmlxxvii, 2024-09-24:
> So I went back to look at my devices, as I use the BME280 with the LilyGo board and have the Heltec board in my test setup.
> (...)
> After spending some time this AM on this, I can only think that there is something off with the Heltec board and I2C. As the exact same code and BME280 worked on the LilyGo and not on the Heltec. I also tried using different pins for I2C, and had the same result.

Thank you very much for the quick turnaround on this matter!

Yeah, I was suspecting there's something odd going on with the Heltec v2 board, and your testing just confirmed that.

I got such suspicion after researching about how to do battery voltage measurement in that board (as I'm using solar to recharge 2 INR 18650 3600mAh batteries in my setup) and found this [discussion thread ](http://community.heltec.cn/t/heltec-wifi-lora-v2-battery-management/147) in Heltec's forum. Apparently they were using pin 21 for signaling a voltage divider for battery voltage reading on pin 13, and then they changed the pins assignment on a later "non-marked" 2.1 board revision... So, kinda messy.

Not sure if that's related or not to this issue though. I guess I'll try using the custom pin assignment in OMG's code to rewire the BME280 on different pins on that Heltec v2 board and test again.

mcmlxxvii, 2024-09-25:
Update to this issue:

It seems Heltec ESP32 Lora boards indeed have sketchy I2C.

During my research on this topic today, I stumbled upon this video that talks about the issue: https://youtu.be/9AObZTDxmPk?si=JUM4mw9zwGZIH6kl

In a nutshell: You need to hookup your I2C sensor to the same pins being used by the OLED Display, which are pin 4 for SDA and pin 15 for SCL in my Heltec v2.1 board.

[image]

The [BME280 sensor breakout board](https://www.amazon.ca/dp/B07KYJNFMD) I'm using already has SDA/SCL pulled-up, so it was just a matter of connecting the wires to the aforementioned pins and then adding _'-DBME280_PIN_SDA=4'_ and _'-DBME280_PIN_SCL=15'_ to the _environments.ini_ file in Gitpod, for building a new binary.

Everything seems to be working as expected now:

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
N:
************* WELCOME TO OpenMQTTGateway **************
N: SYS config not found
N: SSD1306 config initialised
N: Saved SSD1306 config loaded
N: Setup SSD1306 Display end
N: OpenMQTTGateway Version: 20240918-02
N: OTA server cert hash: 02118cbaf5cc02748398d93ecdb0ef2a444c37358a617bfb2206dfe                                             351dc7cf2
N: Config loaded from flash
N: Attempting Wifi connection with saved AP: 0
N: No WebUI config to load
N: OpenMQTTGateway URL: http://[redacted by me]/
N: ZwebUI setup done
N: RF Config not found using default
N: Enable RTL_433 Receiver: 915.00Mhz
N: Setup BME280/BMP280 on address: 0x76
N: Setup BME280/BMP280 on SDA pin: 0x4 ----> Added by me for debugging
N: Setup BME280/BMP280 on SCL pin: 0xF ----> Added by me for debugging
N: Bosch BME280 successfully initialized: 0x60 ----> WOOHOO!
N: ZgatewayRTL_433 setup done
N: Offline disabled
N: OpenMQTTGateway modules: ["HELTEC_SSD1306","WebUI","BME280","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
```

```
N: [ OMG->MQTT ] topic: home/OMG_heltec_rtl_433_ESP_FSK/CLIMAtoMQTT/bme msg: {"tempc":23.7,"tempf":74.66,"hum":59.3877,"pa":98908.3,"altim":203.1366,"altift":666.4587}
```

One last thing:

OMG's documentation about [compatible sensors pinout](https://docs.openmqttgateway.com/setitup/sensors.html#pinout) says that BME280 should be connected to pins 21 for SDA and 22 for SCL (which is not wrong as a general rule). However, Heltec boards seem to be an exception to that...

Wouldn't be beneficial updating the docs to reflect Heltec boards' "particularities"?

Cheers!

NorthernMan54, 2024-09-26:
That pages source is stored here - https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/setitup/sensors.md

you could submit a pull request with the appropriate changes and comments.

the fact that the published heltec pinout is wrong, is frustrating….

closed 2024-09-26 by mcmlxxvii