1technophile/OpenMQTTGateway #2070: Multi AP setups: OMG won't roam to other AP when original gets offline

chris-heo, 2024-09-28:
**Describe the bug**
In multi-AP setups (same SSIDs and Passphrases for all APs), OMG seems to stick to an AP and won't connect to another one if it turns off

**To Reproduce**
Steps to reproduce the behavior:
1. Startup OMG with all APs visible
2. Disconnect power from AP (Ctrl+F "##### AP disconnected from power #####" in log)
3. OMG won't connect to other APs with same SSID, only to previous BSSID
4. Connect AP to power
5. OMG is happy again

**Expected behavior**
re-scan for WiFi instead to "insisting" on last BSSID or reboot after a given number of fails

**Screenshots**
[omg_ap_reconnect.txt](https://github.com/user-attachments/files/17176613/omg_ap_reconnect.txt)

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V1.7.0 with some local mods e.g. static config and #2069)
* Libraries:
```
|-- PubSubClient @ 2.8.0
|-- ArduinoJson @ 6.18.3
|-- ArduinoLog @ 1.0.3+sha.f634509
|-- WiFiManager @ 2.0.14-beta+sha.9f5981e
|-- NimBLE-Arduino @ 1.4.1+sha.b6b8af8
|-- TheengsDecoder @ 1.7.8+sha.6e883d1
|-- Adafruit NeoPixel @ 1.12.2
|-- Adafruit AHTX0 @ 2.0.5
|-- SparkFun BME280 @ 2.0.4
|-- Wire @ 2.0.0
|-- Adafruit Unified Sensor @ 1.1.14
|-- ArduinoOTA @ 2.0.0
|-- DNSServer @ 2.0.0
|-- EEPROM @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Ethernet @ 2.0.0
|-- FS @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- WebServer @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- WiFi @ 2.0.0
|-- Preferences @ 2.0.0
|-- Update @ 2.0.0
```

**Additional context**
* Running on ESP32-C3 mini boards w/  framework-arduinoespressif32 @ 3.20007.0 (2.0.7) & toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
* also seeing weird behavior in similar environment: Multiple APs available and visible during bootup but fails to connect. I will file another ticket for that.

puterboy, 2024-10-02:
Worst case, it should be rebooting after 12 failed attempts to connect to the broker -- and I know that works properly on my setup, sometimes too well :) (in that it seems to lose connection to the broker, then fails to reconnect, then reboots even when the network is up)
Would be helpful if you monitored serial port and shared logs...

chris-heo, 2024-10-03:
I think I found the reason for the behavior - one small detail I forgot to mention:
To avoid manually configuring the gateways, I used the ESPWifiManualSetup, wifi_ssid and wifi_password defines which seem to cause different behavior than using the Wifi Manager UI. 
Skimming through the documentation and User_config.h, I couldn't find any other easy way to set build time wifi credentials, is there any way?

closed 2025-01-17 by github-actions[bot]