1technophile/OpenMQTTGateway #2089: Could the WebUI password be set using build_flags?

Odyno, 2024-10-17:
**Is your feature request related to a problem? Please describe.**
It seems that the default webui password is empty and it is not possible to set it programmatically. This is because it is defined as  `ota_pass`, which in turn is set as `gw_password` in the OTA section... What is the logic behind this?

**Describe the solution you'd like**
I would like to set one default from build_flags with a custom variable like WEBUI_PWD

1technophile, 2024-11-07:
You can set the WebUI password during onboarding with WiFi Manager, this is called the gateway password.

puterboy, 2024-11-28:
I can definitely see the logic for allowing all setup variables to be input via build flags to avoid the necessity of using WiFi Manager -- perhaps with an additional flag to disable WiFi Manager or to only enable WiFi manager if it can't log in to the given gateway.

Perhaps provide a PR if you are interested

closed 2025-03-13 by github-actions[bot]