1technophile/OpenMQTTGateway #2099: Add support for Auriol LD6654

DeckardCain2014, 2024-11-05:
**Is your feature request related to a problem? Please describe.**
Hi, I would like to know if it would be possible to add the Auriol LD6654 weather station support.

**Describe the solution you'd like**
Auriol LD6654 weather station support.

**Describe alternatives you've considered**
Flashed the Lilygo RTL 433 firmware and connected to MQTT. However, my weather station is not discovered.

**Additional context**
--

NorthernMan54, 2024-11-05:
Is it supported by rtl_433?

DeckardCain2014, 2024-11-05:
I don't know, is there a list? Is it a prerequisite for the open mqtt gateway?

DigiH, 2024-11-05:
While there are several Auriol devices supported by rtl_433, your LD6654 weather station does not seem to be included yet

https://github.com/merbanan/rtl_433?tab=readme-ov-file#running

It's not a typo by any chance above, as an -LD5664 is supported? 😉

DeckardCain2014, 2024-11-05:
Basically it's a 4-LD5564 😁

DigiH, 2024-11-05:
So the LD5664 is supported by rtl)433, but disabled by default, as you might have seen on the above linked page, due to

https://github.com/merbanan/rtl_433/blob/master/src/devices/auriol_4ld5661.c#L107 

That might also be the reason why @NorthernMan54 has not ported this particular decoder over to rtl_433_ESP (yet)?!?

DeckardCain2014, 2024-11-06:
Actually it is a 4-LD5564, not a 5664 :D So it might be not at all supported. I think I will return it then...

closed 2024-11-06 by DeckardCain2014