1technophile/OpenMQTTGateway #2111: Help

SerpikoIT, 2024-11-12:
good morning I have an inkbird idt-34c-b thermometer I wanted to know if it can be used with Openmqtt Gateway

DigiH, 2024-11-13:
Hi @SerpikoIT 

As the IDT-34c-B seems to be a Bluetooth BBQ thermometer it really depends on it sending it temperature values through BLE advertising data.

Can you turn on **BT: Publish Advertisement data** in the gateways's HA UI and then post a few MQTT messages, along with which probes are connected and their temperatures.

[image]

If the relevant information is being broadcast in the BLE broadcasts, with your help  we can create a decoder for it, similar to the other BBQ thermometer decoders. But it might also be that the information can only be retrieved by connection to the device.

SerpikoIT, 2024-11-13:
forgive me I don't know where to activate BT: Publish Advertisement data I have HA on raspberry

DigiH, 2024-11-13:
In the HA discovered gateway UI, where all the other gateway settings can be adjusted.

Can't you see it in the **Overview** dashboard?

And just to confirm, you are using OpenMQTTGateway on an ESP32 and not possibly Theengs Gateway (HA Add-n)?

SerpikoIT, 2024-11-13:
yes I just installed theeng gateway and configured it, now what should I do??

DigiH, 2024-11-13:
In the Theengs Gateway configuration page it is under **PUBLISH_ADVDATA**

[image]

Turn it on and monitor the undecoded MQTT messages for your IDT-34c-B with MQTT Explorer and post some sample messages here with the information of which probes were connected and what temperatures in ºC it is showing on the device's display.

SerpikoIT, 2024-11-13:
ok I'm not at home now, this afternoon I'll do everything and send you the images. Thank you

SerpikoIT, 2024-11-13:
Here I was able to find this, I hope it helps
[image]

DigiH, 2024-11-14:
Yes, this is what is required to see if the IDT-34c-B might be broadcasting its tempertures, and if it is possible to create a decoder for it. If you're sure this is the correct MAC address/id selected above.

Could you copy and past any messages, with the copy button next to the **Value** heading

[image]

and then remove some of the probes and wait for the next MQTT message to copy, and take one probe into your hand to heat it up and again wait for the next message to copy.

With the 2-3 more examples it should be obvious, as currently I don't see any immediate similarities in the manufacturerdata with the display image showing almost four identical temperature values.

SerpikoIT, 2024-11-14:
ok I'll try right away

{"id":"A4:C1:38:C6:EE:E7","name":"IDT-34c-B","rssi":-46}

[image]

{"id":"A4:C1:38:C6:EE:E7","name":"IDT-34c-B","rssi":-39}

[image]

{"id":"A4:C1:38:C6:EE:E7","name":"IDT-34c-B","rssi":-41}

[image]

{"id":"A4:C1:38:C6:EE:E7","name":"IDT-34c-B","rssi":-53}

[image]

I hope this goes well

DigiH, 2024-11-14:
This would have been perfect, but the undecoded raw **manufacturerdata** is not included in the messages any longer 😞 

Did you turn off **BT: Publish Advertisement data**?

Any chance of the above replaced with the **manufacturerdata** included?

SerpikoIT, 2024-11-14:
{"manufacturerdata": "1f846266a4c138c6eee700", "name": "IDT-34c-B", "id": "A4:C1:38:C6:EE:E7", "rssi": -36}

[image]

{"manufacturerdata": "1f846266a4c138c6eee700", "name": "IDT-34c-B", "id": "A4:C1:38:C6:EE:E7", "rssi": -40}

[image]

{"manufacturerdata": "1f846266a4c138c6eee700", "name": "IDT-34c-B", "id": "A4:C1:38:C6:EE:E7", "rssi": -33}

[image]

SerpikoIT, 2024-11-14:
MQTT_HOST | MQTT host address, i.e. the IP/hostname of your Moquitto Broker Add-on

should I put the HA IP?

DigiH, 2024-11-14:
> {"manufacturerdata": "**1f846266a4c138c6eee700**", "name": "IDT-34c-B", "id": "A4:C1:38:C6:EE:E7", "rssi": -36}
> {"manufacturerdata": "**1f846266a4c138c6eee700**", "name": "IDT-34c-B", "id": [...]

Unfortunately the manufacturerdata doesn't change with the different probes attached or not and the different temperatures. Therefore it won't currently be possible to create a decoder through the BLE advertising data.

SerpikoIT, 2024-11-14:
ok could you help me understand if the configuration of theengs is done correctly??? Does the HA IP address go to the MQTT Host part?

DigiH, 2024-11-14:
> Does the HA IP address go to the MQTT Host part?

It depends on where you are running your MQTT Broker.

And **GENERAL_PRESENCE** should be off for you using it with HA, and **PRESENCE** also only if you are using a room presence implementation in HA.

**PUBLISH_ALL** and **PUBLISH_ADVDATA** can also be turned off again, now that you have posted the sample messages above.

SerpikoIT, 2024-11-14:
ok thanks I'll do it now, I don't know whether to return my thermometer to Amazon or wait. I saw a guy say he was able to read the values ​​with catttool. I'll post the link so we'll see if he can help us out

https://community.home-assistant.io/t/how-to-tranform-gatttool-commands-into-availible-homeassistant-integration/642344

DigiH, 2024-11-15:
It's very likely that you can get values by connection to the device, but that is out of the scope of Theengs Decoder which currently only decodes freely broadcast BLE advertising data.

You could try the [OpenMQTTGateway READ command](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt), to see if you can read from any appropriate service/characteristics.

SerpikoIT, 2024-11-15:
I returned it to Amazon and then on Black Friday I'll get the other model if I can. Thank you

DigiH, 2024-11-16:
@SerpikoIT Have a look at the already compatible devices/BBQ thermometers at

https://decoder.theengs.io/devices/devices.html

closed 2025-03-01 by github-actions[bot]