1technophile/OpenMQTTGateway #2118: RF support for DSC Security Devices

Nealtron, 2024-11-21:
**Is your feature request related to a problem? Please describe.**
The OpenMQTTGateway project doesn't list "DSC Security Contact" under the list of supported Decoders. I'm requesting that support for wireless "DSC Security Contact" sensors be enabled. 

**Describe the solution you'd like**
Add (or uncomment) the the device encoder protocol number for DSC Security devices within the rtl_433 configuration. I've monitored these devices from my RTL-SDR for years and they appear to use Protocol 23 under the rtl_433 project. I believe that translates to Protocol 34 on rtl_433_ESP.

**Describe alternatives you've considered**
No alternatives that I'm aware of.

**Additional context**
I wouldn't mind trying to test support for the new device type, but I can't seem to determine where to set which protocols are enabled on my device. Reviewing the [documentation](https://docs.openmqttgateway.com/use/rf.html#rtl-433-device-decoders) for rtl_433 device decoders, I can't seem to line up the list of supported decoders with rtl_433_esp project. According to the docs, Protocol 34 is already enabled, but the decoder is described as "Esperanza EWS" instead of "DSC Security Contact". 

OpenMQTTGateway Documentaton:
[image]

rtl_433_ESP documentation:
[image]

DigiH, 2024-11-21:
Hi @Nealtron 

The docs just seem to be outdated, as the doc list is also only 81 lines long, up to decoder 102, while the rtl_433_ESP page has 135 lines for OOK decoders and 74 FSK decoders - so the Esperanza EWS shifted down in the rtl_433_ESP list, and your  DSC Security Contact is now 34.

But the decoder numbers don't really matter, the current release 1.8.0 of OpenMQTTGateway includes the latest rtl_433_ESP library, so all decoders listed should be available already.

If you install an rtl_433 OOK build on whatever compatible board you have you should correctly get the DSC Security Contact correctly decoded. Just keep in mind that the reception of any of the ESP32 boards with whichever different radio receiver might not have as wide a range as your previously used RTL-SDR.

Hope this clarifies things a bit.

Nealtron, 2024-11-21:
Hi DigiH, I appreciate the information. I installed the "heltec-rtl_433" image via the web utility, which I believe is the OOK implementation, since FSK is explicitly identified on the other image.

[image]

[image]

[image]

My OpenMQTTGateway has posted over 20,000 messages to MQTT in the last 24 hours, but it hasn't reported a single DSC sensor. I have approximately 30 DSC sensors in my house with some as close as 3 feet away, so I'm not sure what the issue would be.

DigiH, 2024-11-21:
Hi @Nealtron, yes, the "heltec-rtl_433" image is the correct OOK binary to install.

I assume you have a 433 MHZ version of the Heltex board, and not some 868MHZ or other version, All I can think of now is that the DSC require some tweaking of the receiving frequency in the WebUI. Do you get any hints about the exact frequency when using your RTL-SDR and turning on verbose logging?

Nealtron, 2024-11-21:
@DigiH,

I'm not sure if this is detailed enough, but here's the output from the RTL_SDR:

```
root@XXX-PI02:~ # rtl_433 -M level -M stats -vv -R 23
rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR SoapySDR
Registering protocol [23] "DSC Security Contact"
[Protocols] Registered 1 out of 250 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: RTLSDRBlog, Blog V4, SN: 00000001
Detached kernel driver
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
[dsc_callback] Contact Raw Data
codes     : {40}a32fed1f88
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-11-21 10:07:00
model     : DSC-Security id        : 3140895
closed    : 1            event     : 1             tamper    : 0             Battery   : 1             xactivity : 1             xtamper1  : 0             xtamper2  : 0             exception : 0             esn       : 2fed1f
status    : 163          status_hex: a3            Integrity : CRC
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.8 dB      SNR       : 8.3 dB        Noise     : -9.1 dB
[pulse_slicer_pcm] DSC Security Contact
codes     : {55}fa397fb63f8800
[dsc_callback] Contact Raw Data
codes     : {40}a12fed1f83
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-11-21 10:07:02
model     : DSC-Security id        : 3140895
closed    : 0            event     : 1             tamper    : 0             Battery   : 1             xactivity : 1             xtamper1  : 0             xtamper2  : 0             exception : 0             esn       : 2fed1f
status    : 161          status_hex: a1            Integrity : CRC
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.7 dB      SNR       : 8.5 dB        Noise     : -9.2 dB
[pulse_slicer_pcm] DSC Security Contact
codes     : {58}fa197fb63f83000
[dsc_callback] Contact Raw Data
codes     : {40}a12fed1f83
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-11-21 10:07:02
model     : DSC-Security id        : 3140895
closed    : 0            event     : 1             tamper    : 0             Battery   : 1             xactivity : 1             xtamper1  : 0             xtamper2  : 0             exception : 0             esn       : 2fed1f
status    : 161          status_hex: a1            Integrity : CRC
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.1 dB      SNR       : 9.2 dB        Noise     : -9.3 dB
[pulse_slicer_pcm] DSC Security Contact
codes     : {58}fa197fb63f83000
[dsc_callback] Contact Raw Data
codes     : {40}a12fed1f83
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-11-21 10:07:03
model     : DSC-Security id        : 3140895
closed    : 0            event     : 1             tamper    : 0             Battery   : 1             xactivity : 1             xtamper1  : 0             xtamper2  : 0             exception : 0             esn       : 2fed1f
status    : 161          status_hex: a1            Integrity : CRC
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.1 dB      SNR       : 9.5 dB        Noise     : -9.6 dB
[pulse_slicer_pcm] DSC Security Contact
codes     : {58}fa197fb63f83000
[dsc_callback] Contact Raw Data
codes     : {40}a12fed1f83
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
[... 17 lines cut ...]
channel   : C            Battery   : 1             Temperature: 22.0 C       Humidity  : 56 %          Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 8.0 dB        Noise     : -8.0 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-11-21 09:57:27
model     : DSC-Security id        : 2837696
closed    : 1            event     : 0             tamper    : 0             Battery   : 1             xactivity : 0             xtamper1  : 0             xtamper2  : 0             exception : 0             esn       : 2b4cc0
status    : 195          status_hex: c3            Integrity : CRC
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-11-21 09:57:30
model     : Acurite-6045M id        : 179
channel   : A            Battery   : 0             temperature: 68.9 F       humidity  : 45 %          strike_count: 217         storm_distance: 17        active_mode: 1            rfi_detect: 0             data_exception: 0
raw_message: c0b3af2d50f96cd1d5
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.1 dB      SNR       : 8.5 dB        Noise     : -8.6 dB
```

Nealtron, 2024-11-22:
> What I also do not know is how the **ASK** Modulation might affect your missing devices. @NorthernMan54 will likely have more insight into that.

It appears that the Acurite sensors are also using ASK. Wouldn't they be affected to? 

The only real difference I see is that the DSC sensors use CRC for integrity, and the Acurite's either use Checksum or don't have data integrity at all.

Nealtron, 2024-11-22:
I've also changed the Conifgured Frequency to 434 MHz, and it doesn't appear to have had any affect. 

[image]

NorthernMan54, 2024-11-22:
As the issue is specific to a particular decoder, and not the overall code base ( you said the acurite decoder is working ).  It will take someone with access to the sensor methodically looking into what is happening.  There may be something that needs tuning for that particular device.

1 - Use the rtl_433_esp ook example code, so you can tweak code as needed

2 - determine if the board is actually receiving a signal from the sensor.  Enable raw signal debug, and see if a raw signal is received for the actually device.  This is tricky in a signal rich environment.  This will confirm frequency etc.

3 - once you have a raw signal working, look at the signal data waveform to see if the capture is accurate.  The means looking at the decoder logic, and how it decodes a signal and manually running the code thru.  Some decoders may have debug logic to assist. I have added code to a debugger sometimes to see where or why it is failing.

The steps to take it further are all require the physical device.

Nealtron, 2024-11-26:
I won't have time to try anything before Thanksgiving. 

@NorthernMan54,Regarding the test procedure, I'm concerned about signal isolation as I have a lot of sensors in my house.  Do you know if it would work if I placed the the sensor and the ESP32 in a Faraday bag? Alternatively, would you (or anyone else in the community) be interested in testing the sensor?

NorthernMan54, 2024-11-26:
Omg wouldn’t work in a faraday cage ( wifi ), but the rtl_433_esp example might.

personally I’m away for most of the winter without access to my test setup, so I don’t have the ability to test.

Nealtron, 2024-11-26:
> Omg wouldn’t work in a faraday cage ( wifi ), but the rtl_433_esp example might.

I don't know why, but as soon as you said debugging, in my head I assumed that it meant I'd be collecting data via the USB serial connection. I just assumed the WiFi wouldn't be necessary, and that's on me.

closed 2025-03-11 by github-actions[bot]