1technophile/OpenMQTTGateway #2121: Sensors which do not response for a long time are not marked as "unavailable"

pointhi, 2024-11-24:
**Describe the bug**
When a sensor is no longer available (e.g. due to dead battery) Home Assistant is still reporing the current state as if it is a correct reading.

**To Reproduce**
1. Publish sensor reading
2. Remove e.g. battery from the sensor
3. Home assistant is showing the old reading even days later

**Expected behavior**

Expire sensor readings after some (configurable?) time when no new message arrives. See https://www.home-assistant.io/integrations/sensor.mqtt/#expire_after

**Environment (please complete the following information):**
* OpenMQTTGateway version used: v1.8.0

NorthernMan54, 2024-11-26:
Openmqttgateway does not maintain state for any sensors sending data thru it.  This is an issue for the application receiving the sensor data and not omg.

pointhi, 2024-11-26:
The option `expire_after` I shared here is not a state Openmqttgateway needs to keep but an option which tells HA when to expire messages.

Nealtron, 2024-11-27:
I believe this can be set via MQTT Discovery topics. 

The included [rtl_433_mqtt_hass.py](https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py) auto discovery script provided by [rtl_433](https://github.com/merbanan/rtl_433/) has a flag (-x) that sets a single value for the expiration of all data. 

```
    parser.add_argument("-x", "--expire-after", type=int,
                        dest="expire_after",
                        help="Number of seconds with no updates after which the sensor becomes unavailable")
```

```
    if args.expire_after:
        config["expire_after"] = args.expire_after

    logging.debug(path + ":" + json.dumps(config))

    mqttc.publish(path, json.dumps(config), retain=args.retain)

    return True

```

closed 2025-03-12 by github-actions[bot]