1technophile/OpenMQTTGateway #2138: Shelly BLE Button1 discovery causes a rolling reboot

WillMunns, 2024-12-17:
**Describe the bug**

Using a ESP32-WROOM-32, and a stock build of 1.8, pressing the button on the shelly appears to cause the code to reboot.

Builds of 1.7 work correctly

**To Reproduce**
Install 1.8 firmware
Press button on remote
If viewing on the WebUI, you will be re-asked for your password
If viewing on MQTT:

```
xxxxx/LWT offline
xxxxx/LWT online
xxxxx/RLStoMQTT {"latest_version":"v1.8.0","title":"OpenMQTTGateway","release_url":"https://github.com/1technophile/OpenMQTTGateway/releases/tag/v1.8.0","installed_version":"version_tag","entity_picture":"https://github.com/1technophile/OpenMQTTGateway/raw/development/docs/img/Openmqttgateway_logo_mini_margins.png","release_summary":""}
xxxxx/SYStoMQTT {"uptime":4,"version":"version_tag","mqtt":true,"serial":false,"disc":true,"ohdisc":false,"env":"esp32dev-ble","freemem":71612,"mqttp":"1883","mqtts":false,"mqttv":false,"msgprc":0,"msgblck":0,"msgrcv":25,"maxq":0,"cnt_index":0,"minmem":30396,"tempc":47.22,"freestck":3988,"powermode":-1,"eth":false,"rssi":-72,"SSID":"IoT","BSSID":"xxx","ip":"192.168.2.64","mac":"xxx","modules":["WebUI","BT"]}
xxxxx/BTtoMQTT {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"hasspresence":false,"prestopic":"presence/","presuseuuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","pubuuid4topic":false,"ignoreWBlist":false,"forcepscn":false,"tskstck":4668,"crstck":4528,"enabled":true,"scnct":0,"onlysensors":false,"randommacs":false,"filterConnectable":false,"pubadvdata":false,"presenceawaytimer":120000,"movingtimer":60000}
xxxxx/WebUItoMQTT {"displayMetric":true,"webUISecure":true,"displayQueue":0}
xxxxx/LWT offline
xxxxx/LWT online
[repeat]

```
**Expected behavior**

MQTT to show something like
`
xxxxx/BTtoMQTT/MACADDRESS {"id":"MACADDRESS","name":"SBBT-002C","rssi":-37,"brand":"Shelly","model":"ShellyBLU Button1","model_id":"SBBT-002C","type":"BTN","packet":189,"batt":97,"press":5,"mac":"MACADDRESS"}
`

**Environment (please complete the following information):**

Seen on pre-compiled 1.8 binaries and on esp32dev-ble build using gitpod
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
^Unsure on how to get this

DigiH, 2024-12-17:
Hi @WillMunns 

Did you install 1.6.0 over 1.7.0 without erasing the flash, so as to keep your existing settings?

If so, could you erase the flash before installing 1.8.0  again, as there are a few strange inconsistencies in your stated settings 
• even though you have "`adaptivescan":true`, `interval` and `intervalacts` are still at `55555` and `scanduration 10000`, which would mean that any button presses would only ever be recognised once every minute in a 10 seconds window. With `"adaptivescan":true` working correctly it will recognise such devices as the Shelly Button and adjust these values automatically to support continuous scanning, so that all button presses whenever are being recognised.
• You stated that you expect a result with press":5, while all button device decoders have been unified to use the same press/click encodings now, as described on the Shelly Button page 

https://decoder.theengs.io/devices/SBBT-002C.html

If you are still experiencing this after an **erase flash** fresh install of 1.8.0, someone in the team with an actual Shelly Button will have to investigate this.

WillMunns, 2024-12-18:
I installed over something else completely, messed up my settings so i couldn't log in, erased the flash and then installed 1.8 esp32-lolin32lite-ble-firmware.bin which i could access over http and had this error.

Then I installed 1.8 from a fresh build (no clean flash), still has this error, but that is version with the log

Then I installed 1.7 from the releases page (no clean flash), which now works. with the log line from above.

WillMunns, 2024-12-24:
I updated the firmware on the button and that changed the behavior on 1.7, so I thought I better try 1.8 again.

Still crashes, this is a snip of the serial log if that helps

`N: Scan begin
N: BT Device detected: 4D:6F:4C:46:52:54
N: BT Device detected: 44:60:EC:36:50:8E
N: BT Device detected: DD:9D:5F:E5:92:4C
N: BT Device detected: DC:0F:BF:D3:9A:28
N: BT Device detected: 52:23:71:7E:B6:40
scan number 1 endvice detected: 630:C9:22:82:D2:06 devices, 

N: [ OMG->MQTT ] topic: BLE2MQTT/OMG_LOLIN32LITE_BLE/BTtoMQTT/30C92282D206 msg: {"id":"30:C9:22:82:D2:06","rssi":-89} 
N: Scan begin
N: BT Device detected: 4D:6F:4C:46:52:54
N: BT Device detected: 44:60:EC:36:50:8E
N: BT Device detected: F4:B3:B1:76:83:67
N: Active and continuous scanning required, paramaters adapted
Guru Meditation Error: Core  1 panic'ed (Double exception). 

Core  1 register dump:
PC      : 0x400950fe  PS      : 0x00040836  A0      : 0x800d88ff  A1      : 0x3ffe8c30  
A2      : 0x3ffe8ed8  A3      : 0x3ffe8d79  A4      : 0x0000000a  A5      : 0x00000000  
A6      : 0x3ffe8d7c  A7      : 0x00000001  A8      : 0x40080080  A9      : 0x3ffe8d30  
A10     : 0x00060a36  A11     : 0x00040026  A12     : 0x00000003  A13     : 0x3ffe8d7c  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000002  
EXCVADDR: 0xffffffe0  LBEG    : 0x40091df1  LEND    : 0x40091e01  LCOUNT  : 0xfffffff7  

Backtrace: 0x400950fb:0x3ffe8c30 0x400d88fc:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 0x400950fb:0x3ffe8cf0 0x400950fb:0x3ffe8d10 0x400950fb:0x3ffe8d30 0x4008007d:0x3ffe8c30 0x40125d0a:0x3ffe8c50 0x400950fb:0x3ffe8c90 0x400950fb:0x3ffe8cb0 0x400950fb:0x3ffe8cd0 |<-CONTINUES

ELF file SHA256: b6e79d3911c74000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
N: 
************* WELCOME TO OpenMQTTGateway **************
[   173][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
N: SYS config not found
N: OpenMQTTGateway Version: version_tag
N: OTA server cert hash: 02118cbaf5cc02748398d93ecdb0ef2a444c37358a617bfb2206dfe351dc7cf2
N: Config loaded from flash
N: Attempting Wifi connection with saved AP: 0
N: Attempting Wifi connection with saved AP: 1
N: Attempting Wifi connection with saved AP: 2
E (5146) wifi:sta is connecting, return error
[  2374][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
N: Attempting Wifi connection with saved AP: 3
E (6158) wifi:sta is connecting, return error
[  3385][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
N: Attempting Wifi connection with saved AP: 4
N: Connect your phone to WIFI AP: OMG_ESP32_BLE with PWD: THEPASS
*wm:AutoConnect 
*wm:Connecting to SAVED AP: THESSID
E (8683) wifi:sta is connecting, return error
[  5911][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.2.64
[  7730][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
N: No WebUI config to load
N: OpenMQTTGateway URL: http://192.168.2.64/
N: ZwebUI setup done
[  9235][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
N: BT config not found
N: BLE scans interval: 55555
N: BLE connects interval: 3600000
N: BLE scan duration: 10000
N: Publishing only BLE sensors: false
N: Publishing random MAC devices: false
N: Adaptive BLE scan: true
N: Active BLE scan interval: 55555
N: minrssi: -100
N: Presence Away Timer: 120000
N: Moving Timer: 60000
N: Force passive scan: false
N: Enabled BLE: true
N: ZgatewayBT multicore ESP32 setup done
N: Offline disabled
N: OpenMQTTGateway modules: ["WebUI","BT"]
N: ************** Setup OpenMQTTGateway end **************
N: Reconfiguring MQTT client...
N: Update check, free heap: 94296N: Using memory cert
N: Update check done, free heap: 44580N: Connected to broker
`

closed 2025-04-09 by github-actions[bot]