1technophile/OpenMQTTGateway #2143: ESP8266 build fails using build flags to set MQTT settings

towerofpower256, 2025-01-01:
**Describe the bug**
Build fails when configuring the MQTT server settings using build flags that are strings:
* MQTT_USER
* MQTT_PASS

Produces this error:
```
<command-line>: error: invalid conversion from 'const char*' to 'char' [-fpermissive]
main/User_config.h:295:93: note: in expansion of macro 'MQTT_USER'
  295 |     {ss_server_cert, ss_client_cert, ss_client_key, OTAserver_cert, MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false},
      |                                                                                             ^~~~~~~~~
<command-line>: error: invalid conversion from 'const char*' to 'char' [-fpermissive]
main/User_config.h:295:104: note: in expansion of macro 'MQTT_PASS'
  295 |     {ss_server_cert, ss_client_cert, ss_client_key, OTAserver_cert, MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false},
      |                                                                                                        ^~~~~~~~~
<command-line>: error: invalid conversion from 'const char*' to 'char' [-fpermissive]
main/User_config.h:296:46: note: in expansion of macro 'MQTT_USER'
  296 |     {"", "", "", "", MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false},
      |                                              ^~~~~~~~~
<command-line>: error: invalid conversion from 'const char*' to 'char' [-fpermissive]
main/User_config.h:296:57: note: in expansion of macro 'MQTT_PASS'
  296 |     {"", "", "", "", MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false},
      |                                                         ^~~~~~~~~
<command-line>: error: invalid conversion from 'const char*' to 'char' [-fpermissive]
main/User_config.h:297:46: note: in expansion of macro 'MQTT_USER'
  297 |     {"", "", "", "", MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false}};
      |                                              ^~~~~~~~~
<command-line>: error: invalid conversion from 'const char*' to 'char' [-fpermissive]
main/User_config.h:297:57: note: in expansion of macro 'MQTT_PASS'
  297 |     {"", "", "", "", MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false}};
```

Relating to this code:
```cpp
ss_cnt_parameters cnt_parameters_array[cnt_parameters_array_size] = {
    {ss_server_cert, ss_client_cert, ss_client_key, OTAserver_cert, MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false},
    {"", "", "", "", MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false},
    {"", "", "", "", MQTT_SERVER, MQTT_PORT, MQTT_USER, MQTT_PASS, MQTT_SECURE_DEFAULT, MQTT_CERT_VALIDATE_DEFAULT, false}};
```

The build log output:

```
Processing wemosd1-buildfail (platform: espressif8266@4.2.1; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (4.2.1) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
Converting main.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PicoMQTT @ 1.1.1
|-- ArduinoJson @ 6.18.3
|-- ArduinoLog @ 1.0.3+sha.f634509
|-- WiFiManager @ 2.0.5-beta+sha.8452df7
|-- esp8266_mdns @ 0.0.0-alpha+sha.b7c88fda89
|-- LEDManager
|-- TheengsUtils
|-- Wire @ 1.0
|-- ArduinoOTA @ 1.0
|-- DNSServer @ 1.1.1
[... 78 lines cut ...]
build_flags =
  ${com-esp.build_flags}
  '-DMQTT_SERVER="192.168.1.200"'
  '-DMQTT_PORT=1883'
  '-DMQTT_USER="asd"'
  '-DMQTT_PASS="asd"'
board_build.flash_mode = dout
```
3. Select the test platformio environment.
4. Build the project.
5. Observe the failed build.

**Expected behavior**
The build is expected to complete successfully.

**Environment (please complete the following information):**
* OpenMQTTGateway v1.8.0
* VSCode 1.96.1
* PlatformIO Core v6.1.16 Home v3.4.4

**Additional context**
At current, the build badges appear as "no status". Not sure if there's a bug preventing build that's known.

Screenshot:
[image]

towerofpower256, 2025-01-01:
I think I found the issue.

The build flag `MQTT_PORT` wants to be a **string** but I was setting it as a **number**. 

The issue went away after changing:
```
build_flags =
  '-DMQTT_PORT=1883'
```
to:
```
build_flags =
  '-DMQTT_PORT="1883"'
```

closed 2025-01-01 by towerofpower256

randy197, 2025-05-17:
Thank you for the follow up! I've been chasing this for an hour!! Cheers!