1technophile/OpenMQTTGateway #2145: BRESSER-6IN1 constant errors with home assistant errors (connected by mqtt) - Error parsing value: 'dict object' has no attribute 'battery_ok'

fir3drag0n, 2025-01-02:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I use a Bresser 6-in1 weather station. I use a lilygo (433 mqtt esp device) in order to trasmit the values to home assistant. I am constantly getting these errors, as it seems, the value is not expected

`Logger: homeassistant.helpers.template
Quelle: helpers/template.py:809
Erstmals aufgetreten: 16:37:29 (95 Vorkommnisse)
Zuletzt protokolliert: 17:23:29

Error parsing value: 'dict object' has no attribute 'battery_ok' (value: {"model":"Bresser-6in1","id":810889475,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":202,"rain_mm":2,"startup":1,"flags":1,"mic":"CRC","protocol":"Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A","rssi":-63,"duration":33000}, template: {{ value_json.battery_ok | is_defined }})
Error parsing value: 'dict object' has no attribute 'battery_ok' (value: {"model":"Bresser-6in1","id":810889475,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":202,"rain_mm":2,"startup":1,"flags":1,"mic":"CRC","protocol":"Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A","rssi":-78,"duration":34001}, template: {{ value_json.battery_ok | is_defined }})
Error parsing value: 'dict object' has no attribute 'battery_ok' (value: {"model":"Bresser-6in1","id":810889475,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":202,"rain_mm":2,"startup":1,"flags":1,"mic":"CRC","protocol":"Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A","rssi":-62,"duration":33000}, template: {{ value_json.battery_ok | is_defined }})
Error parsing value: 'dict object' has no attribute 'battery_ok' (value: {"model":"Bresser-6in1","id":810889475,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":202,"rain_mm":2,"startup":1,"flags":1,"mic":"CRC","protocol":"Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A","rssi":-63,"duration":32996}, template: {{ value_json.battery_ok | is_defined }})
Error parsing value: 'dict object' has no attribute 'battery_ok' (value: {"model":"Bresser-6in1","id":810889475,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":202,"rain_mm":2,"startup":1,"flags":1,"mic":"CRC","protocol":"Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A","rssi":-79,"duration":32996}, template: {{ value_json.battery_ok | is_defined }})`

**To Reproduce**
Steps to reproduce the behavior:
1. Connect device to lillygo or similar device, connection by mqtt
2. the device and the entities are created
3. See error log

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V1.8, stable)
* Home Assistant Core 2024.12.5

DigiH, 2025-01-02:
Doesn't the Bresser 6in1 send two different MQTT messages, one with temperature, humidity etc. which also includes the battery_ok value, but also a second kind of message with your above posted wind and rain properties. 

So while both sets should be discovered, which I hope they are for you, it looks as if Home Assistant gets confused with any station sending such alternating MQTT messages?!? Apart from the errors in the log, is any functionality also breaking or working fine?

I have experiences something similar with my Mi Band, but also only in very recent updates of Home Assiatnt, but things are working fine, but the Mi band also produces varying MQTT messages.

DeckardCain2014, 2025-01-13:
I have the same issue. Also, the value for rain changes between a 2-digit and a 5-digit number. This comes with a change in the field 'flags' toggling between 0 and 1.
@DigiH Is there any documentation you are referring to concerning the 'two messages'?

DigiH, 2025-01-13:
> @DigiH Is there any documentation you are referring to concerning the 'two messages'?

Well, from seeing and reading some threads I know that some stations do send separate messages if and when they have very many properties, which would fit into on single message, and for this particular Bresser 6in1 this also seems to be stated in the actual rtl_433 decoder itself ;)

https://github.com/NorthernMan54/rtl_433_ESP/blob/main/src/rtl_433/devices/bresser_6in1.c#L30-L32

But anyone who actually has this station can easily check the individual MQTT messages history with MQTT Explorer or any other MQTT monitoring app to see which kind of messages are being published, and if there are variations in the messages or not.

@DeckardCain2014 Maybe you could post some samples of the different messages.

If and why HA might have changed this to produce error messages if and when some key might be missing I do not know, also not if this is supposed to be a feature with the ability to disable it somewhere, or if it might be a bug in current releases.

DeckardCain2014, 2025-01-13:
```
{"model":"Bresser-6in1","id":70624531,"channel":0,"sensor_type":1,"wind_max_m_s":1.8,"wind_avg_m_s":1.8,"wind_dir_deg":208,"rain_mm":63.2,"startup":1,"flags":1,"mic":"CRC","protocol":"Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A","rssi":-51,"duration":34996}
{"model":"Bresser-6in1","id":70624531,"channel":0,"temperature_C":-2.4,"humidity":86,"sensor_type":1,"wind_max_m_s":1.2,"wind_avg_m_s":1.2,"wind_dir_deg":198,"rain_mm":68917.89,"uv":0,"startup":1,"flags":0,"mic":"CRC","protocol":"Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A","rssi":-52,"duration":34996}
```

DeckardCain2014, 2025-01-13:
no only these two :D

DigiH, 2025-01-13:
So it seems that the `battery_ok` property used to be there, hence the auto-discovery entry for battery_ok, but it doesn't seem to be published at all any longer, hence the error message for the previously discovered battery_ok.

If this is an upstream issue with rtl_433 or something else I could not say, and someone more savvy with the rtl_433 decoders will need to look into this.

In the meantime it might be good if one of you could verify this with rtl_433 and an RTL-SDR stick, to see if the received.decoded and published messages are the same as with rtl_433_ESP and OpenMQTTGateway.

To not receive any more error messages you could always delete homeassistant > binary_sensor > battery_ok discovery entry for this station with MQTT Explorer.

closed 2025-04-29 by github-actions[bot]