1technophile/OpenMQTTGateway #2146: web flash resulted in wrong definition

pauldogg, 2025-01-03:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
flashed a heltec lora 32 using the webflasher https://docs.openmqttgateway.com/upload/web-install.html
Mac os, chrome.

resulted in device definition of: esp32_olm_poe

**To Reproduce**
Steps to reproduce the behavior:
1. use chrome to flash via web
2. connect via hot spot
3. setup wifi
4. wrong device definition
5. device doesn't connect

**Expected behavior**
properly connect to wifi after setting up credentials using the AP

i am able to connect to the ap but once set device will not connect to wifi.

1technophile, 2025-01-03:
Try to clean your cache maybe, I've just installed heltec-wifi-lora-32 successfully from the V1.8.0 web upload

alfista2600, 2025-01-11:
fwiw I had this happen twice while doing a bunch of re-flashing. I figured it was just me skipping a step but it happened twice in a row. Reloading the page worked, but I was confused when I got the wrong firmware despite picking what I expected.

closed 2025-04-27 by github-actions[bot]