1technophile/OpenMQTTGateway #2150: Hostname for the ESP isn't set to the gateway name, but left to the device default

melyux, 2025-01-12:
**Describe the bug**
I would expect the hostname of the OMG device to be set to the Gateway Name we specify. However, it's the default ESP hostname instead (my Gateway Name is "omg-xxxx", but its hostname shows up as "ESP-103F51").

**To Reproduce**
Steps to reproduce the behavior:
1. Set the Gateway Name of the OMG device.
2. Check the hostname of the device on the router.
3. See that it's not the Gateway Name but the ESP's original hostname.

**Expected behavior**
I would expect the Gateway Name to be used as the hostname. I can still ping "omg-xxxx.local" with mDNS, but the hostname registered to the router is "ESP-103F51".

**Environment (please complete the following information):**
* OpenMQTTGateway version used: latest
* Library version related to the problem you have: N/A

melyux, 2025-02-17:
What did you modify?

melyux, 2025-05-19:
No

melyux, 2025-11-17:
@q127981 What did you modify?

1technophile, 2026-01-11:
@claude create a PR for this