1technophile/OpenMQTTGateway #2154: BM2 car voltage ble sensor not detected

fragolinux, 2025-01-19:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
hi, i've a BM2 battery monitor, phone detects it and reads its data, but cannot get them in my HA instance... flashed an esp32 with OMG 1.8.1 esp32dev-ble latest, and it detects lots of other ble devices in my house and around, but not the BM2...

[image]

**To Reproduce**
Steps to reproduce the behavior:

i just put the esp32 about 5 meters away from the BM2

**Expected behavior**

values from my BM2 dected and sent to home assistant

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

as you can see, no signs of any device starting with "5054..."

[image]

**Environment (please complete the following information):**
* OpenMQTTGateway version used 1.8.1 esp32dev-ble edition, flashed via official web installer

**Additional context**
Add any other context about the problem here.

uptime	8923
version	"v1.8.1"
mqtt	true
serial	false
disc	false
ohdisc	false
env	"esp32dev-ble"
freemem	85152
mqttp	"1883"
mqtts	false
mqttv	false
msgprc	1402
msgblck	0
msgrcv	1402
maxq	28
cnt_index	0
minmem	32280
tempc	55
freestck	2868
powermode	-1
eth	false
rssi	-49
SSID	"MYOWNWIFI"
BSSID	"DA:44:89:29:DF:4A"
ip	"192.168.1.47"
mac	"30:AE:A4:01:D0:24"
modules	"'WebUI', 'BT'"
origin	"/SYStoMQTT"

BT	
bleconnect	true
interval	55555
adaptivescan	true
intervalacts	55555
intervalcnct	300000
scanduration	10000
hasspresence	false
prestopic	"presence/"
presuseuuid	false
minrssi	-100
extDecoderEnable	false
extDecoderTopic	"undecoded"
pubuuid4topic	false
ignoreWBlist	false
forcepscn	false
tskstck	1620
crstck	3008
enabled	true
scnct	136
onlysensors	false
randommacs	false
filterConnectable	false
pubadvdata	false
presenceawaytimer	120000
movingtimer	60000

WebUI	
displayMetric	true
webUISecure	true
displayQueue	0

fragolinux, 2025-01-19:
sorry, i forgot to add that i even run this

mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"intervalcnct":300000}'

nothing arrived on my mqtt, yet

DigiH, 2025-01-19:
> i've a BM2 battery monitor, phone detects it and reads its data, but cannot get them in my HA instance

With the BM2 one big issue is that they are being resold by many different companies, in different regions of the world, with many of them implementing their own little speciality in the BLE broadcast, which makes and made it already awkward for us to correctly recognise all of these BM2 variants.

Can you turn on 

[image]

in the gateway's HA UI, and the post some sample MQTT messages from your BM2 which you can view through MQTT Explorer connected to your MQTT broker.

Then we should be able to see what peculiarity your BM2 has in its BLE broadcasts, making it currently unrecognisable.

DigiH, 2025-01-19:
> as you can see, no signs of any device starting with "5054..."

And if you cannot see anything with this Bluetooth MAC address, did you have a look in the app settings, if the BLE broadcasts might be switched off for your BM2? Can you see it with the above turned on?

fragolinux, 2025-01-19:
[image]

i enabled it, nothing appearing in mqtt explorer...

checked the app, no such setting available

device is less than 5m away from the esp32

fragolinux, 2025-01-19:
in case, can you suggest a working device, even different from bm2, but that works for sure? my car was halted from 29 december to last tuesday and the battery went down, a friend had to come with an electric starter... now i bought one for myself, this morning again, after just 4 days, again flat battery, i had problems attaching the clamps and going to turn on the car, I'm disabled, for me the car is vital... i'll buy a more powerful battery, too, car repair said that probably was too small and warranty is now over (3 years old)

DigiH, 2025-01-19:
Then it could well be that your particular BM2 stops sending BLE advertising broadcasts once it is paid with the proprietary app, as some other Bluetooth sensors do as well.

Before deleting it from the app though, to see if it then picks up with BLE advertisement boradccastst again, could you also see with the free nRF Connect app for your phone to see if it picks up any thing for the BM2?

All your OpenMQTTGateway settings look fine, and you can see that you are picking up a lot of other BLE devices, it just appears that your BM2 does not send out any BLE broadcasts at all to be picked up by OMG.

fragolinux, 2025-01-19:
Here the sshots[image]

[image]

[image]

![Screenshot_2025-01-19-16-25-18-269_no.nordicsemi.android.mcp.jpg](https://github.[image]

enshot_2025-01-19-16-23-09-850_no.nordicsemi.android.mcp.jpg …]()

[image]

fragolinux, 2025-01-19:
this just appeared inside mqtt explorer, nothing else

{
  "id": "50:54:7B:80:63:01",
  "mac_type": 0,
  "adv_type": 0,
  "manufacturerdata": "fb525da6d59a095f2fbc6782e30bf107",
  "rssi": -95
}

DigiH, 2025-01-19:
> Here the sshots

Thanks, in the 3rd and 4th screenshot I can see that even the phone has a pretty low reception of the BM2, with its values always in the lower -80 - -100 range.

What kind of ESP32 do you have, where is it plugged in,. possibly still in the PC, where the same 2.4 GHz frequency WiFi and other Bluetooth devices might be preventing it from receiving the BM2? Or possibly close to a WiFi route?

Can you try and plug the ESP32 into a separate USB phone charger, and position it away from any other such possible interference sources?

And while screenshot 4 shows that it also broadcasts a proprietary manufacturer data, screenshot 5 shows it also broadcasts the iBeacon format with which it should be recognisable and decodable for OpenMQTTgateway.

So in your case it not showing up at all, decoded or not, seem to be purely due to a reception issue, likely through the placement of your ESP32.

Remember that with any BM2 mounted on a car battery it is sort of inside a faraday cage, so reception is always a bit tricky, usually remedied with either a different placement of the ESP32 or with an ESP32 with an external antenna.

DigiH, 2025-01-19:
> this just appeared inside mqtt explorer, nothing else
> 
> { "id": "50:54:7B:80:63:01", "mac_type": 0, "adv_type": 0, "manufacturerdata": "fb525da6d59a095f2fbc6782e30bf107", "rssi": -95 }

Yes, this was the proprietary manufacturerdata message I just mentioned above, it suddenly appearing means that your ESP32 luckily got one single BLE broadcast from the BM2, due to its placement. With better placement it should also received the beacon messages which should make OpenMQTTGateway be able to correctly recognise and decode the BM2.

DigiH, 2025-01-19:
You could also try setting the current

`"minrssi": -100,`

to

`"minrssi": -110,`

Not as an ideal solution as a better placement of the ESP32, as receiving such low broadcasts can also result in corrupt receptions.

fragolinux, 2025-01-19:
phone, i took screenshots from about double the distance of the esp32, which is a devkit v1, the normal ones similar to "nodemcu" as shape... 

the esp32 is connected to an usb 2A psu, not at pc... i'll try to move it at about 3m from my car, can't do better than this, unfortunately...

the strange thing is that i can get the bm2 values from phone at this distance, and the esp32 cannot... it's about 2 meter below the access point, i'll move it nearer the car, i'll update asap, thanks

DigiH, 2025-01-19:
> the strange thing is that i can get the bm2 values from phone at this distance, and the esp32 cannot... it's about 2 meter below the access point, i'll move it nearer the car, i'll update asap, thanks

It really depends on what kind of ESP32 it is, but also the particular production quality of you single unit. Phones are usually a lot better with reception, even of low power broadcasts like from BM2s surrounded my metal car bodies.

And with the one single message you did receive above, the "minrssi": -100, shows that it only received this one broadcast which was luckily above the -100 threshold. 

If you have a different ESP32, it is definitely worth testing with that one, as you should received a broadcast at East once every minute with youth settings you posted above, and as you likely do with all the other more powerful devices.

> the esp32, which is a devkit v1, the normal ones similar to "nodemcu" as shape...

I also usually go for the ESP32-WROOM-32D types, as their built in antenna is more free standing that most other dev kit ESP32, as I've already also seen slight differences in reception quality with them, nothing compared to a ESP32-WROOM-32U with a good external antenna connected to its antenna connector, or other external antenna ESP32 options - just FYI 

Also, general RF observations, the higher the ESP32 will be placed, the better its reception will also be. Any power outlet close to the ground is the worst option.

fragolinux, 2025-01-19:
should i turn back any of those parameters modified before? like the {"intervalcnct":300000}?

fragolinux, 2025-01-19:
i have other ones, like the esp32-c3 supermini, is it supported? i can try it, in case... which firmware should i put on it? i just see a "mini", not the supermini, on the upload page

DigiH, 2025-01-19:
> under it i can now see the BM2:

Great that you got it working now with a different placement of the ESP32 :)

I have to admit I don't know HA well enough, but my test setup also has this via Unnamed device, but the MAC addresses do show the correctly recognised devices. Isn't this something new and it should likely have the particular gateway name - something we might need to look into.

> should i turn back any of those parameters modified before? like the {"intervalcnct":300000}?

`intervalcnct` defines with which interval OpenMQTTGAteway connects to the BM2 to fetch the voltage, as this can only be retrieved by connection, whereas the level is being received by passively scanning the BLE broadcasts. As connections do have a small toll on the actual battery lie itself you should keep it ate a reasonable value as not to additionally deplete your car battery even quicker than it already seems to be. So putting it back to its default 3600000 might be better than connecting every 5 minutes.

> i have other ones, like the esp32-c3 supermini, is it supported?

I honestly don't know, as I am not that savvy with all the different board definitions and usually concentrate working on BLE decoding, but you could always try some c3 binaries and see how it works for you.

Also worth reading through 

https://community.openmqttgateway.com/t/shocking-discovery-about-antenna-quality-of-esp32-modules/2228

fragolinux, 2025-01-19:
before it was a doit devkit v1, now as you suggested, a similar one with wroom-32d... same position, NO config edits at all, it already picked my bm2... thanks, much appreciated your kind help! Going to donate to the project :)

{"id":"50:54:7B:80:63:01","mac_type":0,"adv_type":0,"manufacturerdata":"4c000215655f83caae16a10a702e31f30d58dd82f1f24da11f","rssi":-89,"brand":"GENERIC","model":"BM2 Battery Monitor","model_id":"BM2","type":"BATT","track":true,"batt":31,"device":"BM2 Tracker"}

DigiH, 2025-01-19:
Closing this issues than as completed :)

closed 2025-01-19 by DigiH

fragolinux, 2025-01-19:
@DigiH just FYI, i tried an esp32-c3 supermini, and it works, too

{"id":"50:54:7B:80:63:01","name":"Battery Monitor","rssi":-76,"txpower":0,"brand":"GENERIC","model":"BM2 Battery Monitor","model_id":"BM2","type":"BATT","batt":31,"device":"BM2 Tracker"}

any way to have that {"intervalcnct":300000} survive a reboot? it's reset each time i reboot it... i'd like to have it run every 15 minutes...

thanks again for your help!

fragolinux, 2025-01-19:
about the "unnamed" device, i just found this, which seems to be just that, a connector between the bm2 and the esp32 device managing it... strange name, so i just renamed it in HA

[image]

fragolinux, 2025-01-24:
@DigiH oh, i didn't know of not renaming... ok, no problem, i can just remove all the entities, restart and let it recreate as they were originally... i'll try an other one at my parents house, tomorrow, a separate HA instance... and i just saw an update c3 supermini v2 on aliexpress, with embedded rgb led, but most important, external antenna... ordering!