1technophile/OpenMQTTGateway #2164: autodiscovery handles negative integer id wrong

alexxx-k, 2025-02-17:
Context
My sensor is a Bresser 5in1 that uses the 6in1 Protocol by autodetection. I use the prebuild [lilygo-rtl_433-fsk](https://docs.openmqttgateway.com/prerequisites/boards/lilygo-rtl_433-fsk)

Describe the bug
The autodiscovery message is replacing a leading "-" by "/", what renders the sensor to not subscribe to the right topic.

{
"stat_t": "+/+/RTL_433toMQTT/Bresser-6in1/0//2059337613",
"dev_cla": "precipitation",
"unit_of_meas": "mm",
"name": "Rain",
"uniq_id": "Bresser-6in1-0--2059337613-rain_mm",
"val_tpl": "{{ value_json.rain_mm | is_defined }}",
"stat_cla": "measurement",
"device": {
"ids": [
"Bresser-6in1-0--2059337613"
],
"cns": [
[
"mac",
"Bresser-6in1-0--2059337613"
]
],
"mdl": "Bresser-6in1",
"name": "Bresser-6in1-0--2059337613",
"via_device": "OMG_lilygo_rtl_433_ESP_FSK"
}
}

the message is then published to topic:
home/OMG_lilygo_rtl_433_ESP_FSK/RTL_433toMQTT/Bresser-6in1/0/-2059337613

manually fixing the autodiscovery message fixes the issue.
(replacing "stat_t": "+/+/RTL_433toMQTT/Bresser-6in1/0//2059337613" by "stat_t": "+/+/RTL_433toMQTT/Bresser-6in1/0/-2059337613")

Expected behavior
handle negative integer values in ids correctly or maybe the negative value itself is an integer overflow

Embedded Device (please complete the following information):
Running on LilyGo_SSD1306
using [Theengs OpenMQTTGateway v1.8.1]
https://docs.openmqttgateway.com/prerequisites/boards/lilygo-rtl_433-fsk.html

I created an issue here, https://github.com/matthias-bs/BresserWeatherSensorReceiver, because mixed up the projects.
Maybe using the BresserWeatherSensorReceiver could in the OpenMQTTGateway may fix also https://github.com/1technophile/OpenMQTTGateway/issues/2145

DigiH, 2025-02-18:
As we'd never really want a state topic to have a double **//** , best to check after the - > / replace for any such occurrence to then replace again with **/-** !?!

Just a quick stab at a theoretical fix.

@alexxx-k Can you make your own build with this change to test this?

https://github.com/1technophile/OpenMQTTGateway/compare/development...DigiH:OpenMQTTGateway:rtldiscovery

bjv, 2025-05-21:
I have the same issue, reported in #2193 2193

> [@alexxx-k](https://github.com/alexxx-k) Can you make your own build with this change to test this?

unfortunately I am not able to do so, but the code fix looks suitable to my problem as well

DigiH, 2025-05-21:
@bjv 

I'll create a test build for a web upload, which should be available in about 4 hours for all day tomorrow, until it is overwritten by the nightly builds again just after midnight UTC. 

Let us know if it fixes it for you

DigiH, 2025-05-22:
@bjv 

The test build  with **SHA:8085f6** is now available for the next 24 hours from the development web upload page at

https://docs.openmqttgateway.com/dev/upload/web-install.html

alexxx-k, 2025-05-23:
Thanks to you both.

Sorry for late response, i was on a business trip all the week and had no time.
Don't worry about the testbuild, i can build it myself. Just keep the branch.

As soon as i get another board for testing, i will do. In the meanwhile i used it in another project, so i don't have test hardware right now.

Greets
Alex

closed 2025-09-06 by github-actions[bot]