1technophile/OpenMQTTGateway #2170: Major MQTT corruption when reading battery service/char

DigiH, 2025-03-18:
Fresh OMG 1.8.1 installation of [esp32dev-ble](https://docs.openmqttgateway.com/prerequisites/boards/esp32dev-ble) binary. 

**Describe the bug**
trying to read the battery levels of my Playbill Candles again, first trying through my openHAB automation, but then also separate by issuing the READ command(s) through MQTT Explorer. There is predominantly major MQTT message corruption, instead of receiving correct responses on the correct device topic. 

Trying to read the battery level with

**Command topic:**
home/OMG_ESP32_BLE/commands/MQTTtoBT

**JSON playload:**
```
{
	"ble_read_address": "AA:BB:CC:DD:EE:FF",
	"ble_read_service": "180f",
	"ble_read_char": "2a19",
	"value_type": "INT",
	"ttl": 4,
	"immediate": true
}
```

With such strange corrupt MQTT message results, on a topic "d", even though the response JSON does look fine apart from the also "id":"d" ;)

[image]

With the corruption sometimes even going as far as the whole top topic home/internal getting mangled into the general home/OMG_ESP32_BLE/BTtoMQTT list.

This screenshot also shows that this seems to occur mainly with successful responses, whereas failed responses - produced by OMG itself when not receiving any response from the device - look fine, but also only most of the time.

[image]

This should be reproducible with any device which has the standard Bluetooth SIG Battery level 180f/2a19 service/char.

DigiH, 2025-05-12:
@h2zero @1technophile 

Can you reproduce this issue on your end? I am constantly getting it, also with WRITE commands now when trying them with a new device, and also still after updating with the latest https://github.com/1technophile/OpenMQTTGateway/pull/2177

DigiH, 2025-05-14:
Ok, testing it again with a default pre-built binary `esp32dev-ble` things are working fine as expected.

I suspect it is me using all manual definitions for `ESPWifiManualSetup=true` in my custom environment, but likely especially

`NetworkAdvancedSetup=true`

with all its network definitions. Just for complete info, I also have (un)defined

'-UMQTTsetMQTT'
'-UMQTT_HTTPS_FW_UPDATE'
'-DCHECK_OTA_UPDATE=false'

Whatever is messing with the MQTT publishing of the READ and WRITE responses I really don't know, but it is happening all the time for me.

Happy to give any further info on my custom environment required to get to the bottom of this ;) so I can hopefully READ and WRITE again without any hiccups.

DigiH, 2025-05-15:
Hi @1technophile, with both, the 1.8.1 release and with the latest development, both with my manual WiFi, MQTT and Network definition environment, e. g.

```
[env:esp32dev-ble-test]
platform = ${com.esp32_platform}
board = esp32dev
extra_scripts = ${com-esp32.extra_scripts}
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.ble}
  ${libraries.decoder}
build_flags =
  ${com-esp32.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_PIN=2'
  '-DLED_PIN_ON=0'
  '-DGateway_Name="OMG_BLE_TEST"'
  '-DAttemptBLEConnect=false'
  '-DAdaptiveBLEScan=false'
  '-DTimeBtwActive=60000'
  '-DTimeBtwRead=20000'
  '-DpubBLEAdvData=true'
  '-Dgw_password="XXXXXXXX"'
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="SSID"'
  '-Dwifi_password="PASSWORD"'
  '-DNetworkAdvancedSetup=true'
  '-DNET_MASK="255.255.255.0"'
  '-DNET_GW="192.168.0.1"'
  '-DNET_DNS="192.168.0.1"'
  '-DNET_IP="192.168.0.2"'
  '-DMQTT_SERVER="192.168.0.3"'
  '-DMQTT_USER=""'
  '-DMQTT_PASS=""'
  '-DMQTT_PORT="1883"'
  '-UMQTTsetMQTT'
  '-UMQTT_HTTPS_FW_UPDATE'
  '-DCHECK_OTA_UPDATE=false'
```

DigiH, 2025-05-15:
When working on my NOTIFY test here for my power stations a similar corruption started happening.

Changing the BLEdata["id"] assignment from a pointer to a temporary string to a persistent local variable seems to fix the issue - at least for my repeated battery level checking tests for the last half hour.

Only changed the two occurrences at the moment in this test branch for you to comment

https://github.com/1technophile/OpenMQTTGateway/compare/development...DigiH:OpenMQTTGateway:corruption

h2zero, 2025-05-16:
@DigiH nice catch, your change is now storing a copy of the ID instead of a pointer to a local variable, surprised this bug hasn't shown up sooner, maybe due to a change in ArduinoJson? Either way, this is a safe fix.