1technophile/OpenMQTTGateway #2186: Red Led

Sc0th, 2025-05-17:
Can anyone enlighten me as to what they 4th light on the Theengs Bridge indicates when flashing red (I assume something negative!) I am unable to find anything in the docs to help.

The first three LED's are Solid Green, the fourth alternates between 3 blue flashes which I understand is ble scanning, then 3 red flashes which I cannot decipher.

My config looks a little like this 

`blegw/001/SYStoMQTT {"uptime":210073,"version":"v1.8.1","rgbb":255,"mqtt":true,"serial":false,"disc":false,"ohdisc":false,"env":"theengs-bridge-v11","freemem":68008,"mqttp":"1883","mqtts":false,"mqttv":false,"msgprc":125916,"msgblck":0,"msgrcv":125916,"m
axq":28,"cnt_index":0,"minmem":13744,"tempc":53.33,"freestck":3424,"powermode":-1,"eth":false,"rssi":-49,"SSID":"Scothular","BSSID":"20:36:26:D5:45:70","ip":"192.168.63.3","mac":"D0:EF:76:23:F8:68","modules":["WebUI","BT"]}
blegw/001/BTtoMQTT {"bleconnect":false,"interval":5000,"adaptivescan":false,"intervalacts":5000,"intervalcnct":3600000,"scanduration":1000,"hasspresence":false,"prestopic":"presence/","presuseuuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecode
rTopic":"undecoded","pubuuid4topic":false,"ignoreWBlist":false,"forcepscn":false,"tskstck":1620,"crstck":3056,"enabled":true,"scnct":34873,"onlysensors":true,"randommacs":false,"filterConnectable":true,"pubadvdata":false,"presenceawaytimer":120000,"movin
gtimer":60000}`

I am only trying to control 4 switchbot curtain 3 devices, which ~20% of the time do not respond on first message and I have to send multiple in order for them to be actioned, I am guessing this may be linked to whatever the red LED is trying to tell me.

TIA Scoth

1technophile, 2025-05-17:
@Sc0th thanks for your support by purchasing a bridge.

The Red Led means that an error has been detected by the bridge during the processing.

Could you log to the WebUI and go to the console ?

The WebUI is accessible through your bridge IP address, login: admin, password entered during onboarding

[image]

Once in the console, looks for a line starting by E: or W:
You can share an extract to [my email](https://github.com/1technophile) directly if you don't want to share your BLE Mac addresses.

1technophile, 2025-05-17:
Also I would suggest to increase your scan interval to reduce conflict between scans and BLE commands:
```
"interval":5000,"adaptivescan":false,"intervalacts":5000
```
You could go at 50000 for `interval` and `intervalacts`

closed 2025-08-31 by github-actions[bot]