1technophile/OpenMQTTGateway #2205: Trying to follow the docs but most links are broken

sparky672, 2025-06-23:
I am on this page here:

[https://docs.openmqttgateway.com/prerequisites/board.html](https://docs.openmqttgateway.com/prerequisites/board.html)

Simply trying to figure out which ESP32 board to buy and what I would need to install on it.  There are 75 "environments" listed on that page and absolutely no idea how to differentiate many of them.

So then I go to read more:

_"Choosing your board depends heavily on the technologies you want to use with it. To have a good overview of the compatibilities per board you can refer to the compatible modules attributes of each [board (opens new window)](https://compatible.openmqttgateway.com/index.php/boards/)."_

And the link to look at compatible boards is completely dead.  Actually, all of the links to _compatible.openmqttgateway.com_ are dead.

[image]

I had no trouble figuring out, installing, and configuring HomeBridge, but this documentation seems a little more vague.  Can somebody please dumb this down a little more for me?  

I have a bunch of Govee H5074 temperature sensors, which are only BLE (no hubs available) and a working HomeBridge setup, soon to be migrated to a Raspberry Pi 5.  I am thinking of putting an ESP32 closer to the Govee devices to be an extender or intermediary between BLE and Wifi in order to get into HomeBridge running the "homebridge-mqttthing" plugin.  Maybe I don't need it since the BLE devices will be close enough to the Raspberry Pi, but an openmqttgateway setup would seem like a good option to have.

Thank you!

mmiller7, 2025-08-07:
I'm also having this same issue but its certificate errors instead.

Example: [List of supported devices](https://compatible.openmqttgateway.com/index.php/devices/), door/window sensors, PIR sensors, smoke detectors, weather stations...

[image]

sparky672, 2025-11-06:
I've completely abandoned this solution.  

I found that the Govee hub sold with their water/flood sensors also works with their H5074 temperature sensors.  When the hub is configured in the Govee app, you can use it with the Homebridge Govee plugin and everything works as expected.